MarlinFirmware/Marlin #27257: Shift layers when enabled input shaping

Eksol, 2024-07-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In earlier versions of Marlin, such a bug was not observed. After updating the marlin version to 2.1.2.4, a bug appeared that manifests itself in the form of a shift of layers along the Y axis. If you disable input shaping, the bug disappears.

### Version of Marlin Firmware

2.1.2.4

### Printer model

ender 3

### Electronics

BTT V1.3

### LCD/Controller

CR10 display

### Other add-ons

3D-touch, PT1000

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16159405/Marlin.zip)

thisiskeithb, 2024-07-10:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Eksol, 2024-07-10:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Okay!

Eksol, 2024-07-10:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Thanks! Problem solved

closed 2024-07-10 by thisiskeithb