MarlinFirmware/Marlin #27258: [BUG] Compiler Warning on MarlinCore.cpp

fnsign, 2024-07-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

 see screenshot 
 [image]

### Bug Timeline

first appearance ca. 3 weeks ago

### Expected behavior

No compiler warning

### Actual behavior

Compiler warns

### Steps to Reproduce

Start compiler on VSCode

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 Pro

### Electronics

BIGTREETECH SKR MINI E3 V1.2

### LCD/Controller

stock

### Other add-ons

--

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-07-10:
Please attach your configuration files.

This does not occur on stock ramps configs

ellensp, 2024-07-10:
Setting   #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2
and #define SERIAL_PORT 1
is enough to trigger this minor warning.

ellensp, 2024-07-10:
Issue is stm32 implementation of analogInputToDigitalPin (uint32_t)  vs marlins pin definition (int32_t)

ellensp, 2024-07-11:
code already has

#pragma GCC diagnostic ignored "-Wnarrowing"

so add this  line under it

#pragma GCC diagnostic ignored "-Wsign-compare"

Matsavol, 2024-07-11:
I can confirm same problem happening on Longer LK1 (Alfawise U20). Visual Studio Code does compile it showing this error, but does produce bin image. Once the printer is flashed with the resulting image it gets stuck in the "Load EEPROM defaults" screen touchscreen not responding.

ellensp, 2024-07-11:
This warning does not cause any other issues

fnsign, 2024-07-11:
> Please attach your configuration files.
> 
> This does not occur on stock ramps configs

Here are my config files:

[configs.zip](https://github.com/user-attachments/files/16171782/configs.zip)

ellensp, 2024-07-12:
PR created, closing issue

closed 2024-07-12 by ellensp