MarlinFirmware/Marlin #27268: [FR] Change the logic of the FILAMENT_RUNOUT_DISTANCE_MM parameter

MrKuskov, 2024-07-15:
### Is your feature request related to a problem? Please describe.

Since Marlin now supports runout sensors with two sensors, it is necessary to change the logic of the FILAMENT_RUNOUT_DISTANCE_MM parameter

### Describe the feature you want

The following logic must be implemented:

The switching sensor and the motion sensor show the absence of a filament in this case, FILAMENT_RUNOUT_DISTANCE_MM is equal to the distance from the filament sensor to the extruder, so that the entire filament can be consumed.
The switching sensor indicates the presence of a filament, and the motion sensor indicates the absence of motion, in this case FILAMENT_RUNOUT_DISTANCE_MM is equal to the accuracy of the motion sensor reading.

### Additional context

A PTFE tube must be installed between the filament sensor and the extruder to fix the distance between the sensor and the extruder

thisiskeithb, 2024-07-15:
> The following logic must be implemented:
> The switching sensor and the motion sensor show the absence of a filament in this case, 

The switch sensor tracks filament presence and the encoder tracks filament motion.

> FILAMENT_RUNOUT_DISTANCE_MM is equal to the distance from the filament sensor to the extruder, so that the entire filament can be consumed.

Yes, that is what `FILAMENT_RUNOUT_DISTANCE_MM` is for:

https://github.com/MarlinFirmware/Marlin/blob/20a704b1545cec61fde66d76eb7fe4c49f76a185/Marlin/Configuration.h#L1960-L1963

> A PTFE tube must be installed between the filament sensor and the extruder to fix the distance between the sensor and the extruder

You should already be running reverse bowden in a direct drive configuration when a motion-based filament sensor is installed off the hotend or the sensor will not be able to properly track filament motion due to slack from the hotend moving around.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-07-15 by thisiskeithb

MrKuskov, 2024-07-15:
@thisiskeithb 
I have a sensor mounted on the printer frame, a 500mm long ptfe tube is installed between the sensor and the direct extruder, the FILAMENT_RUNOUT_DISTANCE_MM value is 500. I tried to stop the movement of the filament to simulate the entanglement of the filament, but the printing continued anyway