MarlinFirmware/Marlin #27270: [BUG] printing does not stop when the BTT SFS v 2.0 sensor detects no movement of the filament

MrKuskov, 2024-07-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a sensor mounted on the printer frame, a 510mm long ptfe tube is installed between the sensor and the direct extruder, to eliminate sagging of the filament. The FILAMENT_RUNOUT_DISTANCE_MM value is 500. I tried to stop the movement of the filament to simulate the entanglement of the filament, but the printing continued anyway.

pins_BTT_SKR_V1_4.h
```
// Filament Runout Sensor
//
#define FIL_RUNOUT_PIN                     P1_26  // E0DET sensor switch from runout
#define FIL_RUNOUT2_PIN                    P1_25  // E1DET sensor motion from runout

```

configuration.h
```
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS  1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.

  #define FIL_RUNOUT_STATE     LOW        // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.

 
  // Commands to execute on filament runout.
  // With multiple runout sensors use the %c placeholder for the current tool in commands (e.g., "M600 T%c")
  // NOTE: After 'M412 H1' the host handles filament runout and this script does not apply.
  #define FILAMENT_RUNOUT_SCRIPT "M600"

  // After a runout is detected, continue printing this length of filament
  // before executing the runout script. Useful for a sensor at the end of
  // a feed tube. Requires 4 bytes SRAM per sensor, plus 4 bytes overhead.
  #define FILAMENT_RUNOUT_DISTANCE_MM 500

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
    // Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    #define FILAMENT_MOTION_SENSOR

    #if ENABLED(FILAMENT_MOTION_SENSOR)
      #define FILAMENT_SWITCH_AND_MOTION
      #if ENABLED(FILAMENT_SWITCH_AND_MOTION)
        #define NUM_MOTION_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_MOTION#_PIN for each.
        #define FIL_MOTION1_PIN        P1_25  // motion switch  (E1DET)

      #endif
    #endif
  #endif
#endif
```

### Expected behavior

I expected the printing to stop

### Actual behavior

printing continued, it is possible that the accuracy of the motion sensor is missing, which is 3mm, plus an error between the inner diameter of the ptfe tube and the diameter of the rod, approximately 1 mm.

### Steps to Reproduce

to simulate the stop of the rod, I pressed the pressure roller in the extruder

### Version of Marlin Firmware

bugfix 2.1.x 07/13/24

### Electronics

skr 1.4

### LCD/Controller

tft 35

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-07-15:
> Steps to Reproduce
> to simulate the stop of the rod, I pressed the pressure roller in the extruder

Don’t just try to hold the filament. Cut it after the sensor so there is no movement.

If that doesn’t work, then there is something wrong with your sensor because I also have the SFS V2 and it properly triggers if there is no motion or runout is detected.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord

closed 2024-07-15 by thisiskeithb

MrKuskov, 2024-07-16:
@thisiskeithb 
Everything seems to be working. But now, when printing starts, the printer prints exactly as much as definitely #define FILAMENT_RUNOUT_DISTANCE_MM 300. Perhaps this is due to the tension of the thread in the PTFE tube. How can I disable thread checking at the beginning of printing?