MarlinFirmware/Marlin #27273: [BUG] (bug summary)

classicrocker883, 2024-07-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Okay I am not sure if this is a bug with negative implications, or it is just whatever it is.

So everything compiles fine, no errors or warnings to note otherwise.
However, when I go to this file:

**Marlin\src\module\stepper.cpp**: line ~530
```
constexpr uint32_t timer_setup_ns = cycles_to_ns(timer_read_add_and_store_cycles);
constexpr hal_timer_t PULSE_HIGH_TICK_COUNT = ns_to_pulse_timer_ticks(_min_pulse_high_ns - _MIN(_min_pulse_high_ns, timer_setup_ns));
constexpr hal_timer_t PULSE_LOW_TICK_COUNT = ns_to_pulse_timer_ticks(_min_pulse_low_ns - _MIN(_min_pulse_low_ns, timer_setup_ns));
```

I get red squiggles under these 3 `constexpr` and this error:

```r
expression must have a constant value
stepper.cpp(528, 37): the value of variable "SystemCoreClock" (declared at line 51 of "\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include\system_stm32f1xx.h") cannot be used as a constant
stepper.cpp(529, 47): expression cannot be interpreted
stepper.cpp(530, 46): expression cannot be interpreted
```

removing `constexpr` from each 3 will remove this error and also compile successfully.

### Expected behavior

Expect no warnings/errors

### Actual behavior

Compiles correctly.
Everything else seems to be fine,
except this error appears.

### Steps to Reproduce

1. Load STM32 configs ([Ender-3V2](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2/CrealityV422/CrealityUI))
2. `STM32F103RE_creality` in **platformio.ini**
3. Build
4. navigate to **Marlin\src\module\stepper.cpp**
5. warnings should popup under "PROBLEMS" tab / red squiggles appear under code in question

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-08-19:
> So everything compiles fine, no errors or warnings to note otherwise.

Closing since there are no compile error or warnings.

closed 2024-08-19 by thisiskeithb

thisiskeithb, 2024-08-19:
@classicrocker883: Please only open bug reports for **confirmed bugs**.

If you're unsure, please [hop on our Discord](https://discord.com/servers/marlin-firmware-461605380783472640) and ask for help before proceeding.