MarlinFirmware/Marlin #27278: [BUG] Ramps 1.4 + Panelolu2 + Serial2 - Incorrect Serial Port 2 pin conflict error

KevWal, 2024-07-16:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

After setting "#define SERIAL_PORT_2 2" in Configuration.h Marlin 2.1.2.4 doesn't compile due to the error "#error "Serial Port 2 pin D16 and/or D17 conflicts with another pin on the board."" from SanityCheck.h.

This is due to the setting of "#define LCD_PINS_EN          EXP1_03_PIN" in pins_RAMPS.h, even though Pin 17 (#define EXP1_03_PIN                    AUX4_17  // 17) is not used by my #define LCD_I2C_PANELOLU2 LCD (see https://web.archive.org/web/20201125221424im_/http://1.bp.blogspot.com/-hy4fhgAV3gc/UXcPlS0FsFI/AAAAAAAAAWk/zxEO8xD9eCY/s640/Ramps_Adapter_Schematic.png)

Commenting out #define LCD_PINS_EN          EXP1_03_PIN in the "      #if ANY(MKS_12864OLED, MKS_12864OLED_SSD1306)" section of pins_RAMPS.h allows compilation and all appears to work fine.

### Bug Timeline

Noticed 16/07/2024

### Expected behavior

Expected successful compilation

### Actual behavior

Failed compilation

### Steps to Reproduce

#define LCD_I2C_PANELOLU2
#define SERIAL_PORT_2 2

### Version of Marlin Firmware

2.1.2.4

### Printer model

GC I3 - https://web.archive.org/web/20160411113355/http://www.emakershop.com/browse/listing?l=964

### Electronics

RAMPS 1.4

### LCD/Controller

Panelolu2

### Bed Leveling

None

### Your Slicer

Slic3r

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/16255517/Config.zip)

ellensp, 2024-07-17:
It was a trap to draw owners of this lcd_i2c_panelolu2 display out of the woodwork (no not really, just a potentially useful coincidence) 

If you would be so kind, please take hires pictures of the front and back of this display and attach them so they can be added to  
https://marlinfw.org/docs/hardware/controllers.html#lcd_i2c_panelolu2

I have confirmed this issue still exists in bugfix 2.1.x

I have created a PR to fix this.

closed 2024-07-17 by thinkyhead

KevWal, 2024-07-19:
Thank you so much for fixing, pictures attached.
[image]
[image]

ellensp, 2024-07-19:
I have added these photos  it to an existing PR that adds another display https://github.com/MarlinFirmware/MarlinDocumentation/pull/546