MarlinFirmware/Marlin #27287: H-Bot or not?

FlailAway, 2024-07-17:
### Is your feature request related to a problem? Please describe.

I cannot find a way to control an H-Bot type design.

### Are you looking for hardware support?

Nope, Marlin Firmware.

### Describe the feature you want

We are out in the boonies and landline phone only and cannot subscribe to Discord etc as no required cell number. Grrrr.

I bought a surplus/used high-precision machine (not a printer, just the X/Y-movement) that I thought was CoreXY. Once here it seems to be a modified H-BOT. It does not have the belt passing over the carriage. The cross over belt+idlers are at the motor end. I am not sure if that translates to the standard H-BOT for the actual movements.

Moving the motors by hand.

Left-motor CW and right-motor CCW moves Y towards zero

Left-motor CCW and right-motor CCW moves X towards zero.

The contra movements moves the axis towards their maximum.

Is there (or can I get) a method within Marlin 2 for me to use this?

thisiskeithb, 2024-07-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-07-17 by thisiskeithb

FlailAway, 2024-07-17:
Thanks Keith, perhaps you missed the "land-line" thing. Did you read my post all the way through?

A land-line is a telephone system using copper wires on poles and it does voice only. It does not do data and cannot be used to verify a new Subscriber to most of the items you copied and pasted. We farmers **feeding you City dwellers,** live without Social Media and cell phones, not by choice but lack of availability.

My posting **is** a "Feature request" and I explained what is needed.