MarlinFirmware/Marlin #27290: EEPROM data size ERROR (Tronxy CXY V10.1 220921 BOARD)

Nixdorf2k, 2024-07-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

so.
I have Tronxy X5SA with board CXY-466-V10.1-220921 based on GD32F427ZET6.
it looks simular to ST32F446.
I try to buid marlin firmware, an everythink goes good. All motor/extruder/nozle/bed/ works fine.
But I can't store configurating to eeprom. Command M500 gives error.
Cli says "EEPROM ERROR Version missmatch and EEPROM ERROR datasize.
[Configuration.zip](https://github.com/user-attachments/files/16313509/Configuration.zip)
attach configuration
and second trouble with screen
[image]

### Version of Marlin Firmware

BugFix2.1.X

### Printer model

Tronxy X5SA

### Electronics

CXY-466-V10.1-220921

### LCD/Controller

Tronxy display

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-07-19:
CXY-466-V10.1-220921 based on GD32F427ZET6. is not a board that is supported in Marlin at this time
We only have the older CXY-446-V10-220413/CXY-V6-191121 (STM32F446ZE)

Nixdorf2k, 2024-07-19:
> CXY-466-V10.1-220921 based on GD32F427ZET6. is not a board that is supported in stock Marlin at this time We only have the older CXY-446-V10-220413/CXY-V6-191121 (STM32F446ZE) Wich is not the same, and has a very different chip

The original Tronxy Marlin seems to work well. 
If you have any idea to resolve this problem, i can try it.

Valerich1971, 2024-07-20:
Greetings, friends. I join the topic starter, I have the same 3D printer with the same board and with the same problems

closed 2024-11-04 by github-actions[bot]