MarlinFirmware/Marlin #27291: [FR] Custom GCode function - Guidance Needed

SaKiEQ, 2024-07-19:
### Is your feature request related to a problem? Please describe.

I am trying to implement a non standard Gcode to extend the functionality of one of my machines. 

So far  I have added the switch condition in:  gcode.cpp  -> case 93: G93(); break;

and extended the gcode.h -> static void G93(); 

Also added the g93.h file containing the function... -> void GcodeSuite::G93() {}
Also all the  #includes to the custom g93.h file.

Somehow the compiler keeps puking, it seems it does not like that I extend the GcodeSuite class. 
Where am I going wrong, why cant  I expand the GcodeSuite  ? 

I cant think of any place to look / or ask other then here. 

Thanks in advance ! 
This is the compiler error: 
```
/var/folders/dc/q_mr0jy11wlgxx_500ywpkn80000gp/T//ccW9Z1Ou.ltrans2.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':
/var/folders/dc/q_mr0jy11wlgxx_500ywpkn80000gp/T/arduino_build_573106/sketch/src/gcode/gcode.cpp:462: undefined reference to `GcodeSuite::G93()'
```

### Describe the feature you want

I just need to create a 'dummy' Gcode to house some custom functions.

ellensp, 2024-07-20:
You need to add the code G93.cpp not G93.h   

 Here is a minimal example

```DIFF
diff --git a/Marlin/src/gcode/gcode.cpp b/Marlin/src/gcode/gcode.cpp
index 11b02ded68..7af3dbcbc2 100644
--- a/Marlin/src/gcode/gcode.cpp
+++ b/Marlin/src/gcode/gcode.cpp
@@ -461,6 +461,8 @@ void GcodeSuite::process_parsed_command(const bool no_ok/*=false*/) {
 
       case 92: G92(); break;                                      // G92: Set current axis position(s)
 
+      case 93: G93(); break;                                      // G93: Custom gcode
+
       #if ENABLED(CALIBRATION_GCODE)
         case 425: G425(); break;                                  // G425: Perform calibration with calibration cube
       #endif
diff --git a/Marlin/src/gcode/gcode.h b/Marlin/src/gcode/gcode.h
index bb2f009994..9437031ab8 100644
--- a/Marlin/src/gcode/gcode.h
+++ b/Marlin/src/gcode/gcode.h
@@ -615,6 +615,8 @@ private:
 
   static void G92();
 
+  static void G93();
+
   #if ENABLED(CALIBRATION_GCODE)
     static void G425();
   #endif
diff --git a/platformio.ini b/platformio.ini
index 76200cbbd5..8365de05d3 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -107,6 +107,7 @@ default_src_filter = +<src/*> -<src/config> -<src/tests>
   +<src/gcode/control/M120_M121.cpp>
   +<src/gcode/control/M999.cpp>
   +<src/gcode/geometry/G92.cpp>
+  +<src/gcode/geometry/G93.cpp>
   +<src/gcode/host/M110.cpp>
   +<src/gcode/host/M114.cpp>
   +<src/gcode/host/M118.cpp>

```

And add the file Marlin/src/gcode/geometry/G93.cpp

```CPP
#include "../gcode.h"

/**
 * G93:
 */
void GcodeSuite::G93() {
}

```

closed 2024-07-20 by ellensp

SaKiEQ, 2024-07-20:
Hi, thanks. I think I did pretty much all of the above, including the .cpp and not .h. 

Only thing I copied my G93.h into motion/G93.cpp not geometry, but that should have not made a difference ? 

Also I am still on Arduino IDE, despite the speed, it still would not link - so I think something is cached somewhere 
that does not resolve with restarts of the machine nor IDE> 

I will try tomorrow to get VScode / platformIO running, so I have more feedback and control.

ellensp, 2024-07-20:
This example builds under Arduino ide also. 
The G93.cpp can be anywhere

SaKiEQ, 2024-07-20:
Thank you very much again. I have been compiling Marlin for years on Arduino IDE, so I knew i would work. Hence I was so puzzled when this small change just would not work - despite my logical thinking. 

Hence I thought I reached out, since I might have missed a larger concept that I might not be familiar with...

ellensp, 2024-07-20:
The arduino ide does seem to have caching issues 
How to force it to clear the cache https://forum.arduino.cc/t/cache-clear-option/1040106/7

SaKiEQ, 2024-07-20:
Thanks again. I just compiled in Arduino IDE. I figured out the cache locations according to the links before, let it run over night (or at least while I slept), since without cache it took way longer to compile the first time around ! 

The funny thing, both my version that originally would not work and you version compile equally.... 

Now off to learn VScode and PlaformIO !

SaKiEQ, 2024-07-20:
Just for context, I am working on integrating my pendant creations. For this I needed a temporary protocol debugging means - I am doing that via a GCODE command so I can send cmds outside of the final serial interface (which will become a secondary interface on the Arduino Mega, piggy backing into the Joystick implementation). 
The larger pendant has a touch screen for function commands apart from jog. The small one just X/Y/Z buttons, Jog and TFT display.
[image]