MarlinFirmware/Marlin #27294: [BUG] Z Offset / Z Home Height disparity

classicrocker883, 2024-07-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Home position + Z Offset does not make sense.

When Z Offset is set to > or < 0.00, the resulting Home height should not be affected, at the very least, the logic is backwards.

Say you have a bed leveling probe, and you set the Z Offset to -0.60, and have `Z_CLEARANCE_DEPLOY_PROBE` = 10. Z Home position will be 10.60.

Likewise, if it were set to +0.60, Z Home position will be 9.40.

Here is a crude demonstration representation of how this looks:

[image]

The blue line indicates when the Nozzle/Hotend is (Z= 0.00)

In this example, *(Middle of image)* let's say Z Offset = 0.00, and the position of the nozzle is Z=0.00, this results in the nozzle (blue line) slightly floating above the bed surface (indicated in grey).  
To correct the difference so the nozzle (blue line) now can touch the bed surface , you change Z Offset = -0.10.  
But as you can see, this increases the Home height by that difference. *(Right side of image)*

Then when you change Z Offset = +0.10, this obviously lets the nozzle float even higher above the bed surface (blue line), yet the Home position is now decreased *(Left side of image).

### Expected behavior

Home position should be 10.00

### Actual behavior

with Z Offset @ -0.60, Home position is 10.60
with Z Offset @ +0.60, Home position is 9.40

### Steps to Reproduce

1. Using UBL, but probably any bed leveling type should do.
2. Go to your printer and Home the machine.
3. If you have set your Z Offset, you may notice the Z Home height saying +/- the Z Offset

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs](https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2%20Neo/CrealityV422/Configuration.h)

DerAndere1, 2024-07-20:
There are multiple z offsets. Which one are you talking about? How did you set it?

ThomasToka, 2024-07-21:
Z-offset: distance from Probe to the nozzle. Z-offset can't be positive cause in this case the nozzle would touch the bed before the probe.

Thats my understanding of the Z-offset.

DerAndere1, 2024-07-21:
Lets assume 
1) you do not print with a workspace offset active
2) //#define HOTEND_OFFSET_Z { 0.0 }
3) the bed surface is meant to be at machine position z=0.
4) your bed leveling mesh matches the physical bed geometry of the heated bed
5) You enable bed leveling compensation after homing

Then, NOZZLE_TO_PROBE_OFFSET is set correctly when the current Z position is reported to be 0.0 after the following Gcode is run:
```
M190; heat the bed
G28; home
G1 Z0; move to z=0.0
```

If you want that the machine moves to a certain position at the end of G28, enable and set Z_AFTER_HOMING accordingly.  
If that does not work as expected, please provide steps to reproduce the issue, including a description how you measure the "Z Home position" (e.g. 10.10)

DerAndere1, 2024-07-22:
This is probably caused by the following line:

https://github.com/MarlinFirmware/Marlin/blob/cc122bec8293f5c49c7a07f5a90a9c0448a48262/Marlin/src/module/motion.cpp#L728

which was introduced in commit https://github.com/MarlinFirmware/Marlin/commit/f53217b2c5bf7d61d21ee6784ec251bc7a670eb3, probably to reach the specified clearance below the probe between probing runs. Maybe @thinkyhead remebers why this was done.

Can we safely change https://github.com/MarlinFirmware/Marlin/blob/cc122bec8293f5c49c7a07f5a90a9c0448a48262/Marlin/src/module/motion.cpp#L742-L745

to this ?
```
  void do_move_after_z_homing() {
    DEBUG_SECTION(mzah, "do_move_after_z_homing", DEBUGGING(LEVELING));
    #ifdef Z_POST_CLEARANCE
      do_z_clearance(Z_POST_CLEARANCE, false, true);
```

thinkyhead, 2024-07-23:
> Z-offset: distance from Probe to the nozzle. Z-offset can't be positive cause in this case the nozzle would touch the bed before the probe.

@ThomasToka — It can indeed be positive, for machines that have the nozzle press into the bed, pushing on a spring mechanism that then triggers a switch, as on the Micromake D1 deltabot, for example.

thinkyhead, 2024-07-23:
The `Z_CLEARANCE_DEPLOY_PROBE` is clearance above Z0 sometimes, but when used in a scenario where the probe will be deployed it refers to clearance above the height at which the probe will get triggered (or above 0, whichever is greater), which inversely corresponds to the probe offset set by `M851 Z<offset>`.

Example: The command `M851 Z-2` indicates that the probe trigger-point is 2mm below the nozzle, so the nozzle will need to be > 2mm above the bed when the probe triggers.

Now, consider the scenario where you physically adjust your probe and issue a new `M851 Z` command. You will want the Z axis to do a different raise for probing in reponse to your change…. Maybe it now needs to be > 3mm above the bed.

JKZuchel, 2024-08-07:
Hi I would like to comment that a got a GEEETECH i3 pro b printer and the first thing I did was flashing Marlin Bugfix 2.1.x and the auto home Z function from the LCD display would do its thing and zero out the Z axis all good there, but when it was done, it would move the extruder back to home and place it 5mm above the printing bed, the issue is that the display would say 5mm but it in reality was 10mm, now that I came across this post, flashed the printer with Marlin Bugfix 2.0.x and now it works just fine.

PSA: Sorry if am not being clear english is my second language and am I new to the 3d printer world.

closed 2024-11-20 by github-actions[bot]