MarlinFirmware/Marlin #27295: [BUG] problems installing marlin 2.1.2.4 on arduino uno on arduino ide

goofyseeker311, 2024-07-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

arduino ide 2.3.2 compiles marlin 2.1.2.4 just fine in the "mega or mega 2560" mode, but when uploading there are time outs, even after closing all 3d printer software. the source code release does not seem to support compilation to arduino avr of arduino uno r3.

### Expected behavior

any arduino avr board should work without problems.

### Actual behavior

refuses to compile to arduino uno and time out when uploading mega or mega 2560 to arduino uno.

### Steps to Reproduce

compile marlin 2.1.2.4 for arduino avr mega or mega 2560 for arduino uno.

### Version of Marlin Firmware

2.1.2.4

### Printer model

custom diy

### Electronics

Arduino Uno R3

### LCD/Controller

none

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/16321507/Configuration.zip)

thisiskeithb, 2024-07-20:
Marlin does not run on an Arduino Uno.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-07-20 by thisiskeithb

ellensp, 2024-07-20:
Uploading is not a function of Marlin.

standard program avrdude is used to send data to the bootloader. 

Marlin will not fit on a Uno, insufficient RAM, FLASH and IO pins 

This is completely outside of marlins control.

goofyseeker311, 2024-07-20:
I just did what the arduino IDE install page said.

thisiskeithb, 2024-07-20:
> cant seem to see a list of the supported devices in official documentation.

Supported boards can be found in [`boards.h`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/core/boards.h) as well as our [Boards list on marlinfw.org](https://marlinfw.org/docs/hardware/boards.html#boards-list).

Please use the links in my [previous comment](https://github.com/MarlinFirmware/Marlin/issues/27295#issuecomment-2241183826) for further help & troubleshooting.