MarlinFirmware/Marlin #27296: [BUG] CALIBRATION_GCODE (Bug or problem)

studiodyne, 2024-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

[Marlin.zip](https://github.com/user-attachments/files/16322622/Marlin.zip)

### Version of Marlin Firmware

bf2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-07-21:
Closing since you have not provided any useful debug information and this is not actionable.

closed 2024-07-21 by thisiskeithb

studiodyne, 2024-07-21:
> Closing since you have not provided any useful debug information and this is not actionable.

Hi , now it's done , take a look if you have , one minute
Thks

studiodyne, 2024-07-21:
how to re open ?

thisiskeithb, 2024-07-21:
> Hi , now it's done , take a look if you have , one minute

We created an Issue template for a reason. Don't create a blank/fake issue just so you can strip everything out and bypass all of the required fields.

Please create a new, proper [Bug Report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) and be sure to fill out the entire form before submitting it.

studiodyne, 2024-07-21:
You win ! I surrender