MarlinFirmware/Marlin #27297: [BUG] ( CALIBRATION_GCODE)  Calibration abort if cube >= 10mm

studiodyne, 2024-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi.
Calibration is buggy or have some little problems to adjust.

Behavior :  If cube >= 10mm ,  G425v or G425T not working , but G425 works.

Reason: The sequence of calibration , have two 'Uncertainty distances'. With this value, are trigged two feedrates (a slow and a fast) and two security Z max moves. The problem is if the cube > 10mm and the security max z move is  ' 5mm' (50mm for the higher uncertainty  ),  the z probing stops because too long distance, calibrations stops... But in G425, the uncertainty is defined to the max value, the calibrations sequences are double , one fast , one slow , and , the security is 50mm for the first probing , and  5mm for the second and calibration can continue.

`  const feedRate_t mms = fast ? MMM_TO_MMS(CALIBRATION_FEEDRATE_FAST) : MMM_TO_MMS(CALIBRATION_FEEDRATE_SLOW);  
`
`  const float limit    = fast ? 50 : 5;`
Solution: 
- The best choice , is to add a double sequence for all calibration command , one fast , one slow. Because of the cnc planes that can be XZ or YZ, all calibrations sequences must be the same for all axis.
- May be , to tweak the max security distance for each 'uncertainty ' with the dimension of the cube in the calculation 

I'm working on it
Thks

Note : Unused 
`#ifndef CALIBRATION_MEASUREMENT_CERTAIN
  #define CALIBRATION_MEASUREMENT_CERTAIN   0.5 // mm
#endif`

### Version of Marlin Firmware

bf2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16322706/Marlin.zip)

thisiskeithb, 2024-07-21:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/27296

thisiskeithb, 2024-07-21:
We created an Issue template for a reason. Don't create a blank/fake issue just so you can strip everything out and bypass all of the required fields.

Please create a new, proper [Bug Report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) and be sure to fill out the entire form before submitting it.

closed 2024-07-21 by thisiskeithb