MarlinFirmware/Marlin #27298: [BUG] (bug summary)

qulet, 2024-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Aktywowałem Funkcje lasera
Fire pulse z menu LCD działa. Laser aktywuję się z gcode M3 SXX.   
uruchomienie takiego kodu :
 LightBurn 1.0.06
; Marlin device profile, current position
; Bounds: X-23 Y-23.5 to X23 Y23.5
G21
G91
; Cut @ 500 mm/min, 50% power
M9
M05
G0X23 Y-23.5 F0
; Layer C00
G1X-46 F500 I S50
G1 Y47 
G1X46 
G1 Y-47 
M9
M05
; return to starting pos
G0 X-23 Y23.5 F0
  nie zapala Lasera -  głowica kreśli prostokąt 46x47 mm bez aktywowanego lasera z mocą 50%

### Bug Timeline

Nowy  

### Actual behavior

Głowica zzapalonym  laserem kreśli prostokąt 46x47 mm z  mocy 50%

### Steps to Reproduce

Uruchomić Marlin 
W menu LCD  aktywować  laser.
z karty lub przez port USB  uruchomić  kod.

### Version of Marlin Firmware

2.1.24

### Printer model

Core XY

### Electronics

MKSSbase

### LCD/Controller

RepRapDiscount_Full_Graphic_Smart_Controller

### Other add-ons

NEJE 40W N40630 CNC głowica laserowa 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16326144/Marlin.zip)

qulet, 2024-07-21:
Przepraszam pomyliłem rzeczywiste zachowanie  ze spodziewanym...

thisiskeithb, 2024-07-21:
Please translate your comments to English so we can better help you. I've updated your issue text & reply.

---

From your config:

> ```c
> #define CONFIGURATION_H_VERSION 02010204
> ```

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

qulet, 2024-07-23:
The laser in Marlin V 2.1.1 works properly.

qulet, 2024-07-23:
Changing the software version fixed the problem.

closed 2024-07-23 by qulet