MarlinFirmware/Marlin #27313: [BUG] Homing axis with endstop already triggered

romanr9999, 2024-07-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If an endstop is already triggered before homing process starts, Marlin does not move away from the endstop and this might lead to bad positioning and even printhead crashing onto bedplate (as it happens in my case) after homing axis.

[image]

[Configuration.zip](https://github.com/user-attachments/files/16384222/Configuration.zip)

### Expected behavior

I'd expect the axis to first move to move away in the direction opposite to the homing direction until the endstop is cleared before starting the homing process.

Same thing should happen with the other axis if any of the endstops is triggered before homing, not just the Z axis.

### Actual behavior

My endstops are inductive sensors. They're positioned so they trigger at the correct position based on the axis direction when homing. In the Z axis case the sensor senses a metal part at 25mm from the bedplate and it stays triggered from 0 to 25mm since the sensed metal part it's placed vertically fixed on it's top to the gantry.

The thing is if I try to home while the sensor is triggered (Z= somewhere between 0 to 25mm) it goes up the value defined in configuration to go upa fter homing (in my case 10mm) and set's the axis as homed in that position. No matter if during that up movement of 10mm the sensor never went open. 
If the  endstop is triggered when homing starts, the axis ends homed with wrong values and printhead tries to go further down from the bedplate.

For example. let's say the Z axis is at 5mm. If I home the Z axis, it goes up 10mm to 15 and it sets that point as 25mm, leaving a 10mm positioning error . Since the sensor is already triggered it never goes down.

### Version of Marlin Firmware

2.1.2.2

### Printer model

Custom made

### Electronics

RAMPS 1.4 + MEGA2560

### LCD/Controller

SSD1306 128x64 Oled

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-07-26:
Homing does not work like that, it homes until triggered, if it is already triggered, job is already done.

Change #define HOMING_BUMP_MM      { 5, 5, 2 }     to values grater than you endstop can trigger ie > 25

Then on home the second home moves to an un triggered position and then homes again.

closed 2024-07-26 by ellensp

romanr9999, 2024-07-26:
HOMING_BUMP_MM > 25 would make homing process take a looooot of time each time.

I know homing doesn't work that way, but maybe it should.

"if it is already triggered, job is already done" if job's already done they why does it move at all?? It's not following that logic either.