MarlinFirmware/Marlin #27315: [BUG] MKS_WIFI_MODULE causing gradual layer shifting

andrademf, 2024-07-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

## Description

I've started struggling with gradual layer shifting, as shown in the pictures below, first notice on ringing tower test:
[image]  

## Troubleshooting
I've started with this checklist:
- Machine parameters: reducing acceleration and speed
- Mechanical smoothness: lubing linear rails, adjusting belt tension, and replacing stepper motors
- Hardware: replacing TMC2209 drivers, increasing current, and even checking for power supply glitches
- Slicer: changing layer height, layer width, and so forth

## Conclusion
None of those previous attempts fixed the issue, so I realized it could be related to my recent addition of the MKS WIFI Module to the printer, which turned out to be the root cause. After disabling it, the issue disappeared.

As an additional note, for consistency, I've been using the same gcode in all tests, and re-enabling the MKS Wifi brings the issue back, showing consistency in the bad behavior.

Also, the layer shift is clearly noticeable on my own design part, which is less aggressive on printer movements compared to the ringing tower:

[image]

## Suggestions

For sure, the MKS guys will be the most knowledgeable on that fix (@mks-viva @MKS-Sean et al), but what I was able to validate is, if the `wifi_looping` is not called when `WORKING`, the printing goes well and the layer shift doesn't happen.

Based on that, I would say there is some concurrency or overload caused by it, because even when the module is disconnected from the board, only with the configuration enabled is enough to cause the problem.

Here the _rough_ code change I've made to validate where the issue came from:
[image]
(Certainly, that won’t be the most adequate solution, as this completely disable the WiFi when printing)

### Bug Timeline

After enabling MKS_WIFI_MODULE

### Expected behavior

The MKS Wifi do what is expected to do, the problem is the side effect it's been causing on the printer movement.

### Actual behavior

The printer is losing few steps every layer causing a linear inclination on the every part printed

### Version of Marlin Firmware

2.1.1

### Printer model

Two Trees SP5

### Electronics

MKS Robin Nano V3.1

### Other add-ons

MKS Wifi

### Bed Leveling

No Bed Leveling

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

## GCode
Used for test, available in two different layer height, 0.15mm and 0.3mm: [gcode_oven_grill_support.zip](https://github.com/user-attachments/files/16392768/gcode_oven_grill_support.zip)

## 3D Printer Specs
Brand: Two Trees 
Model: SP5 (Core XY with MKS Robin Nano V3.1 and MKS WiFi)
Slicer: Orca Slicer 2.1.1

[Configuration.zip](https://github.com/user-attachments/files/16392797/Configuration.zip)

ellensp, 2024-07-26:
Provided configs don't match provided information, they are stock default configs from bugfix 2.1.x

andrademf, 2024-07-26:
@ellensp My bad, I switched branches and didn’t notice that when zipping the files.
Just uploaded the correct files.

andrademf, 2024-07-26:
@ellensp Unfortunately, I could not make the printer work with that version (ad60cdd03067877bfa453a0b18b2dd044802b55e). The printer hangs when printing from the SD card right after homing and stops at the origin. Sadly, printing via serial works but does not replicate the same previous scenario.

andrademf, 2024-07-26:
I’ve done a detailed review, even before opening this thread, and I could not find any significant change in the folder `Marlin\src\lcd\extui\mks_ui\wifi_module.h`, only minor refactorings on the code

andrademf, 2024-07-28:
@ellensp You got it, I've carefully review the changes on the bugfix-2.1.x branch under the folder `Marlin\src\lcd\extui\mks_ui\`,  even without any relevant changes on that path the issue was not replicable with the bugfix branch.
So I'm going to close this issue

closed 2024-07-28 by andrademf