MarlinFirmware/Marlin #27316: [BUG] MKS_MINI_12864_V3 no image without SD card inserted

Am0k-GIT, 2024-07-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On a homemade board with STM32F401RCT6, the MKS_MINI_12864_V3 display does not display anything if the SD card is removed.  If the card is inserted, the image appears. I'm using SPI 2 with pins remapped and I'm guessing it's a matter of SPI initialization being correct. However, I don't know how to properly fix the pins file.

stm32f4.ini:

```
[env:Falcon_v1]
extends                     = stm32_variant
platform_packages           = ${stm_flash_drive.platform_packages}
board                       = genericSTM32F407VET6
board_build.variant         = MARLIN_F4x7Vx
board_build.offset          = 0x0000
board_upload.offset_address = 0x08000000
build_flags                 = ${stm32_variant.build_flags} ${stm32f4_I2C1.build_flags}
                               -Os -DHAL_PCD_MODULE_ENABLED
                               -DHAL_UART_MODULE_ENABLED
                               -DSERIAL_RX_BUFFER_SIZE=1024 -DSERIAL_TX_BUFFER_SIZE=1024
                               -DSTEP_TIMER_IRQ_PRIO=0
monitor_speed               = 115200

[env:Falcon_v1_dfu]
extends                     = env:Falcon_v1
upload_protocol             = dfu

[env:Falcon_v1_stlink]
extends                     = env:Falcon_v1
upload_protocol             = stlink
```

pins file:

```
/* Pin Assignments:
 *
 * A0  TM2_CH1 PWM (HOTBED / TTL / 0-10V) | B0  ADC8 (T_HOTBED)                 | C0  X_DIR                             | D0  (N/A)
 * A1  TM2_CH2 PWM (HOTEND)               | B1  ADC9 (T_HOTEND)                 | C1  X_STEP                            | D1  (N/A)
 * A2  TM2_CH3 PWM (FAN0 / FLOOD)         | B2  PS_ON (pulldown)                | C2  Y_DIR                             | D2  (EXP 2-08) SD_DET
 * A3  TM2_CH4 PWM (FAN1 / MIST)          | B3  (EXP 1-03) LCD_D7               | C3  Y_STEP                            | D3  (N/A)
 * A4  (EXP 2-07) SD_SS                   | B4  A_MIN                           | C4  Z_DIR                             | D4  (N/A)
 * A5  (EXP 2-09) SCK                     | B5  Z_MIN                           | C5  Z_STEP                            | D5  (N/A)
 * A6  (EXP 2-10) MISO                    | B6  (EXP 1-07) LCD_RS / SCL / TX    | C6  A_DIR                             | D6  (N/A)
 * A7  (EXP 2-05) MOSI                    | B7  (EXP 1-08) LCD_EN / SDA / RX    | C7  A_STEP                            | D7  (N/A)
 * A8  Enable motors                      | B8  X_MIN                           | C8  B_DIR                             | D8  (N/A)
 * A9  TX                                 | B9  Y_MIN                           | C9  B_STEP                            | D9  (N/A)
 * A10 RX                                 | B10 PROBE_IN (servo0)               | C10 (EXP 1-08) BEEP / CYCLE_START     | D10 (N/A)
 * A11 USB D-                             | B11 (N/A)                           | C11 (EXP 1-08) LCD_D5 / FEED_HOLD     | D11 (N/A)
 * A12 USB D+                             | B12 PROBE_OUT (Z_PROBE)             | C12 (EXP 1-08) LCD_D6                 | D12 (N/A)
 * A13 SWDIO                              | B13 FILAMENT / DOOR                 | C13 (EXP 1-09) BTN_ENC                | D13 (N/A)
 * A14 SWCLK                              | B14 B_MIN                           | C14 (EXP 2-08) BTN_EN1 / SPINDEL_DIR  | D14 (N/A)
 * A15 (EXP 1-06) LCD_D4                  | B15 (EXP 1-08) KILL / RESET         | C15 (EXP 2-06) BTN_EN2 / SPINDEL_EN   | D15 (N/A)
 */
 
#if NOT_TARGET(STM32F4)
  #error "Oops! Select an STM32F4 board in 'Tools > Board.'"
#elif HOTENDS > 2 || E_STEPPERS > 2
  #error "Too many HOTENDs."
#endif

#define DEFAULT_MACHINE_NAME "FALCON"

//
// Limit Switches
//
   
#define X_STOP_PIN      PE15
[... 256 lines cut ...]
DIY STM32F401RC controller

### LCD/Controller

MKS_MINI_V3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/16402253/Configuration.zip)

Am0k-GIT, 2024-07-28:
Ssory, this is correct pin assigments:

```
/* Pin Assignments:
 *
 * A0  TM2_CH1 PWM          | B0  NC                   | C0  NC                   | D0  CAN_RX               | E0  NC                   |
 * A1  TM2_CH2 PWM          | B1  NC                   | C1  NC                   | D1  CAN_TX               | E1  NC                   |
 * A2  TM2_CH3 PWM          | B2  BOOT 1               | C2  PT1000/2             | D2  NC                   | E2  NC                   |
 * A3  TM2_CH4 PWM          | B3  X_DIR                | C3  PT1000/1             | D3  X_STEP               | E3  NC                   |
 * A4  NC                   | B4  Y_DIR                | C4  TR/2                 | D4  Y_STEP               | E4  NC                   |
 * A5  NC                   | B5  Z_DIR                | C5  TR/1                 | D5  Z_STEP               | E5  TM9_CH1 PWM          |
 * A6  TM3_CH1 PWM          | B6  A_DIR                | C6  KILL                 | D6  A_STEP               | E6  TM9_CH2 PWM          | 
 * A7  TM3_CH1 PWM          | B7  B_DIR                | C7  SD_DET               | D7  B_STEP               | E7  PS_ON                |
 * A8  NC                   | B8  SCL                  | C8  ENC                  | D8  L-RX/TX              | E8  PROBE_IN             |
 * A9  NC                   | B9  SDA                  | C9  BEEP                 | D9  L_RX/RX              | E9  PROBE_OUT            |
 * A10 NC                   | B10 EN2                  | C10 DRIVE_TX             | D10 L_D5                 | E10 FILAMENT             |
 * A11 USB D-               | B11 EN1                  | C11 DRIVE_RX             | D11 L_D4                 | E11 B_MIN                |
 * A12 USB D+               | B12 SS                   | C12 ENA_MOTOR            | D12 L_D7                 | E12 A_MIN                |
 * A13 SWDIO                | B13 SCK                  | C13 NC                   | D13 L_D6                 | E13 Z_MIN                |
 * A14 SWCLK                | B14 MISO                 | C14 OSC2_IN              | D14 NC                   | E14 Y_MIN                |
 * A15 NC                   | B15 MOSI                 | C15 OSC2_OUT             | D15 NC                   | E15 X_MIN                |
 */
```

classicrocker883, 2024-08-01:
something isn't right here, why does this line have `EITHER`?

```y
#if EITHER(FYSETC_MINI_12864_1_2, FYSETC_MINI_12864_2_0)
```

all these instances were replaced with `ANY`

thisiskeithb, 2024-08-01:
> all these instances were replaced with `ANY`

Only in `bugfix-2.1.x`. OP is using `2.1.2.4`:

> Version of Marlin Firmware
> 2.1.2.4

> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

@Am0k-GIT: Your code doesn't work in `bugfix-2.1.x`, so that's not possible.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-08-01 by thisiskeithb