MarlinFirmware/Marlin #27318: [BUG] MKS TFT_LVGL_UI does not support relative extrusion

andrademf, 2024-07-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

## Bug

TFT_LVGL_UI MKS implementation has its own filament runout logic, and using that combined with relative extrusion generates issues when resuming the printing.

My main guess for the root cause of the issue is the usage of G90/G91 to lift the Z axis when going to the park position, causing the running gcode to lose its references for extrusion when resuming.

[image]
(OrcaSlicer option to used relative extrusion)

### Expected behavior

Printing interruption might not change current movement configuration

### Actual behavior

The extruder runs backwards when resume the printing, in some cases even ejecting the filament from the extruder

### Steps to Reproduce

1. Setup a printer with TFT_LVGL_UI
2. Use relative extrusion on the slicer
3. Start a print
4. Trigger the filament runout sensor
5. Restablish the filament sensor
6. Press "Resume"

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Two Trees SP5

### Electronics

MKS Robin Nano V3.1

### LCD/Controller

MKS TFT 3.5

### Other add-ons

MKS Wifi

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/16405092/Configuration.zip)

[gcodes.zip](https://github.com/user-attachments/files/16405094/gcodes.zip)

## Suggestion

Instead of using relative movement to lift the nozzle before moving, the current Z position, which is available right above, can be used to calculate the lifted height. In this way, it is not requried to use G90/G91, which causes the side effect of also resetting the relative extrusion defined at the beggining of the gcode.

[image]

thisiskeithb, 2024-07-28:
MKS' LVGL UI needs *a lot* of work in all kinds of areas, but we have an open PR that will address filament runout:

- https://github.com/MarlinFirmware/Marlin/pull/22595

closed 2024-07-28 by thisiskeithb

andrademf, 2024-07-28:
Thanks @thisiskeithb!