MarlinFirmware/Marlin #27326: [BUG] Dual-Z works bad on GT2560A+

giuliosaoncella, 2024-08-01:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Dual Z Stepper works bad on Geeetech GT2560 rev-A+ board

### Bug Timeline

At first attempt with dual Z stepper upgrade

### Expected behavior

When auto home / auto align I expect:
- Z Axys lifts up a bit
- XY Homing
- Head moves to bed-center
- Z axys lift up a bit again
- BLTouch deploy the probe
- Z Axys gently goes down to perform Z probe / set Z "zero"
- Head moves to specified probe point to perform Z alignemen...

### Actual behavior

- Z Axys lifts up a bit ... OK !!
- XY Homing ... OK !!
- Head moves to bed-center ... OK !!
- Z axys lift up a bit again ... OK !!
- BLTouch deploy the probe ... OK !!
- Z Axys gently goes down to perform Z probe / set Z "zero" ... NOW the bug cause the following:

... while the LEFT (Z1) stepper correctly "pulls down" the Z axys, the RIGHT one (Z2) continues "pushing it UP", causing a XZ mechanical skewing...
... the SAME occours if you try to simply MOVE UP and the MOVE DOWN the Z axys

It seems like if Z2 DIR PIN wasn't setted up correctly

### Steps to Reproduce

As in previous sections, just try to home / align or move z-axys up then down...

### Version of Marlin Firmware

2.1.2.4

### Printer model

Geeetech Prusa I3 Pro W

### Electronics

Stock GT2560 rev.A+ (with servo plug for BLTouch)

### LCD/Controller

Standard LCD2004 / encoder

### Other add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[GT2560_Dual_Z_Marlin.zip](https://github.com/user-attachments/files/16461755/GT2560_Dual_Z_Marlin.zip)

narno2202, 2024-08-02:
@giuliosaoncella , you need to uncomment `#define INVERT_Z2_VS_Z_DIR` in `Configuration_adv.h` as Z1 and Z2 seem to turn in the opposite direction.

giuliosaoncella, 2024-08-02:
Hi, narno2202. I already tried this, but motors AREN'T turing in opposite directions.... Z2 goes ALWAYS CW even when CCW is needed... uncommenting that line the problem "inverts", meaning that the Z axys goes DOWN correctly and Z1 vs Z2 turns in opposite direction when it's time to go up... BTW, this test ensures that Z2 DIR pin is correctly triggered, but seems to have a "inverted logic", if I can say it so... anyway, thanks for the help.

giuliosaoncella, 2024-08-02:
...forget about my last considerations about the "inverted logic"... the truth is that the Z2 DIR pin seems to be triggered in the same way (always HIGH or always LOW) in both directions (CW / CCW): I'm not a expert programmer, but it seems like a "copy and paste" mistake or a bad setup of pin definitions (a "0" instead of a "1" in some place...), but not a simple "invert"... If someone have a idea, please let me know, thanks

narno2202, 2024-08-02:
How are your Z motors connected to the mainboard? on E1 stepper or with a doubler.

thisiskeithb, 2024-08-02:
From your config:

> `#define CONFIGURATION_H_VERSION 02010204`

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, swap your "bad" Z driver with X and see if the issue follows the driver.

giuliosaoncella, 2024-08-02:
(for narno2202) ...As in the comments of Configuration.h and ConfigurationAdv.h I attached the Z2 motor to E1 plug... in the config, Z2 DRIVER TYPE I selected A4988 stepper driver, so causing the first free extruder available (E1, since E0 is already in use and defined as A4988) to act as the second axys motor for Z... the Z2 motor was originally connected to the onboard Z-doubler: I detached from this plug and attached to E1...

giuliosaoncella, 2024-08-02:
(for thisiskeithb)... driver swapping already tested, but same result. About bugfix: I'm going to test it, probably this night, then I'll update about the results... thanks.

giuliosaoncella, 2024-08-02:
OK, bugfix version tested but the problem's still alive and kicky !! ... I'm runnig out of ideas...  :-(

thisiskeithb, 2024-08-02:
While still using the latest `bugfix-2.1.x`, replace `/Marlin/src/pins/mega/pins_GT2560_REV_A_PLUS.h` with the one from this zip: [pins_GT2560_REV_A.h.zip](https://github.com/user-attachments/files/16475592/pins_GT2560_REV_A.h.zip), recompile & reflash.

Swap the X & Z2 motor wires on your controller so Z2 motor wire plugs into X and X plugs into Z2.

What behavior do you see when you move Z up and down? What behavior do you see when moving X left & right?

giuliosaoncella, 2024-08-02:
Ah, ok, now I've understood what You meaned in your previous reply: swap the WIRINGS, not the stepper drivers... anyway, the problem only "mixed up" with this behaviors (manual movements):
Z UP : Z1moves UP / X moves LEFT
Z DOWN : Z1 moves DOWN / X moves LEFT (again !!)
X RIGHT : Z2 moves DOWN
X LEFT: Z2 moves UP
...and this is exactly what I expected... unfortunately...

thisiskeithb, 2024-08-02:
> Ah, ok, now I've understood what You meaned in your previous reply: swap the WIRINGS, not the stepper drivers... 

No, you were supposed to swap the physical drivers in that case.

giuliosaoncella, 2024-08-02:
Yes, this time it was clear to me and tested, but as You can see the results aren't good...
...maybe a HARDWARE issue ???... I mean, E1 DIR PIN alway LOW or HIGH ??? ... or incorrect pin definition for E1 ?...

thisiskeithb, 2024-08-02:
I swapped X & E1 (Z2) pins in the pins file, which is why you had to swap X & E1 (Z2) wiring.

>  Z UP : Z1moves UP / X moves LEFT

This would mean that your board is not flashing, you did not swap X & E1/Z2 wiring, or something is seriously wrong with your wiring/board.

giuliosaoncella, 2024-08-02:
well, I'm quite sure that wiring is correct... I think the board correctly flashing, or at least it seems so from the Arduino IDE and, as I said, I swapped X with Z2 (E1) and in fact the behaviors changed accordingly with original ones... I'll try to wire up an LED on E1 DIR pin and upload a classic BLINK sketch to verify if the state of the pin changes or not... if not, then ot's a hardware issue and board replacement should solve...

giuliosaoncella, 2024-08-02:
#define E1_DIR_PIN                            47

... or, more simply, I'll measure it using a multimeter...

thisiskeithb, 2024-08-02:
If you connect to your motherboard via USB and send `M115` via a program like Pronterface, what does it report?

If you followed the steps listed in https://github.com/MarlinFirmware/Marlin/issues/27326#issuecomment-2266183526 and replaced the pins file, **including swapping your X & E1/Z2 wiring**, and reflashed, then X should not have moved if you commanded Z to move / Z should not have moved if you commanded X to move.

giuliosaoncella, 2024-08-03:
please forgive me, Keith... I'm very tired this week and probably I made something wrong last night... today I toke a veeeery deep breath, relaxed a bit and started over all the testings... these are the report:
-replaced the PINS file with the one You swapped X/E1 pins (probably I replaced the original one with a copy of the same file)
-swapped X with Z2 E1 cables (so now X is connected to E1 plug and Z2 is connected to X plug)
-flashed the board... it seems ok
-manual move X axys FW and BW: the carriage ONLY MOVES LEFT
-manual move Z axys UP and DN: the axys moves UP and DN correctly

-M115 reports:

Cap:MULTI_VOLUME:0
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:1
Cap:LFN_WRITE:0
Cap:CUSTOM_FIRMWARE_UPLOAD:0
Cap:EXTENDED_M20:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:EP_BABYSTEP:0
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
Cap:CONFIG_EXPORT:0
ok

So, probably the issue is a HARDWARE failure on E1 (actually X) DIR pin 47...

I'll preform the measurements I mentiuoned above and then I'll be back... in the meanwhile, thanks for your both help and patience...

giuliosaoncella, 2024-08-03:
...OMG !!... it really seems a pin's failure: whatever I set it LOW or HIGH, it always mantain a 3.8-3.9 voltage level... now I'll try to jumperwire the A4988 stepper driver DIR pin to a unused board pin (maybe Z-MAX endstop, since I don't use it)... once this test was done I hope to came back for closing the issue... see You later.

closed 2024-08-03 by thisiskeithb