MarlinFirmware/Marlin #27327: [BUG] New tests file results in `Permission denied`

classicrocker883, 2024-08-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am attempting to create a new tests file for use with **ci-build-tests.yml** workflows. Such as STM32F103RC_creality.

I can confirm STM32F103RE_creality works (and of course all other present tests), but why not anything new?

there must be something I'm missing, or maybe it's to do with PlatformIO and what boards are available? 

I added that default_envs to the yml, is there another which needs it as well? 

```r
./buildroot/bin//run_tests: line 70: /home/runner/work/Marlin/Marlin/buildroot/bin/../tests/STM32F103RC_creality: Permission denied
make[1]: *** [Makefile:48: tests-single-local] Error 126
make[1]: Leaving directory '/home/runner/work/Marlin/Marlin'
make: *** [Makefile:44: tests-single-ci] Error 2
```

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-08-01:
You didn't make the test executable.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

thisiskeithb, 2024-08-01:
> and how do you do that?

Please use the resources linked in my previous comment or ask your favorite search engine how to make files executable.

closed 2024-08-01 by thisiskeithb