MarlinFirmware/Marlin #27331: [BUG] (bug summary)

dotslashsuperstar, 2024-08-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When Running M119 Bltouch z probe is always TRIGGERED plugged into Bltouch port.  The probe works as if everything is fine and prints nicely.  On the docs it says Bltouch should never display TRIGGERED and it means error state.  

On Marlin 2.1.2.4 the z min endstop is TRIGGERED as well when my z min endstop is removed completely.  It displays open for z min endstop and TRIGGERED for probe with endstop plugged in but not pushed closed so I am guessing there are some components on the endstop affecting the value.  I tried inverting the z min endstop and it displayed open for z min endstop with the endstop removed but bltouch was still TRIGGERED.

On bugfix the z min is removed from M119 but z probe is still TRIGGERED and it prints fine.  

I am not sure if this is a bug or me misconfiguring it.  

Here is what I changed from default Ender 3 V2 on bugfix...
```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
//#define PROBE_MANUALLY
#define BLTOUCH
#define NOZZLE_TO_PROBE_OFFSET { -44, -6, -2.6 }
#define Z_MIN_PROBE_REPEATABILITY_TEST
#define X_MAX_POS 245
//#define AUTO_BED_LEVELING_BILINEAR
#define AUTO_BED_LEVELING_UBL
#define ENABLE_LEVELING_AFTER_G28
#define DEBUG_LEVELING_FEATURE
  #define G26_MESH_VALIDATION
  #define MESH_EDIT_MENU        // Add a menu to edit mesh points
#define Z_SAFE_HOMING
//#define DWIN_MARLINUI_PORTRAIT        // MarlinUI (portrait orientation)
#define DWIN_MARLINUI_LANDSCAPE     // MarlinUI (landscape orientation)
```

And Advanced...
```
#define ADVANCED_PAUSE_FEATURE
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      250  // (mm) The length of filament for a complete unload.
```

### Bug Timeline

?

### Expected behavior

Bltouch open

### Actual behavior

TRIGGERED 

### Version of Marlin Firmware

Marlin-2.1.2.4 and bugfix

### Printer model

Creality Ender 3 v2

### Electronics

4.2.2 Stock Mainboard

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I ran M119 before and after a g28 with logging enabled.

[marlin.zip](https://github.com/user-attachments/files/16489116/marlin.zip)

thisiskeithb, 2024-08-04:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2024-08-04 by thisiskeithb

dotslashsuperstar, 2024-08-04:
Yeah I searched over those sites and didn't see anyone with this issue.  I thought since it was TRIGGERED when it was supposed to be open it would be considered a bug unless my configs are misconfigured.