MarlinFirmware/Marlin #27332: [BUG] do_blocking_move_to_xyzijkuv_w() and do_blocking_move_to_w typos in motion.h

thebjtfellow, 2024-08-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

This is a simple typo problem in motion.h when it HAS_W_AXIS. The affected lines are #393 and #394. 

```
#if HAS_W_AXIS
void do_blocking_move_to_w(const float rw, const feedRate_t &fr_mm_s=0.0f);
void do_blocking_move_to_xyzijkuv_w(const xyze_pos_t &raw, const float w, const feedRate_t &fr_mm_s=0.0f);
#endif
```
in the first function, feedRate_t should not be passed by reference
in the second function, const float w should be const_float_t w

Corrected, it would be: 
```
#if HAS_W_AXIS
void do_blocking_move_to_w(const float rw, const feedRate_t fr_mm_s=0.0f);
void do_blocking_move_to_xyzijkuv_w(const xyze_pos_t &raw, const_float_t w, const feedRate_t fr_mm_s=0.0f);
#endif
```

### Bug Timeline

I think it's new, as most people probably don't use the W axis

### Expected behavior

I expected it to compile, but it didn't. Once I made the changes I noted above, it compiled as I expected it to.

### Actual behavior

It failed to compile from a failed to disambiguate error 

### Steps to Reproduce

Define a drive type for the W Axis
Make sure it has its pins assigned. For me, it was the Octopus Pro V1.1 pin file that needed to be modified
You have to have all the default_axis_steps and all those things properly defined
It will ultimately fail to compile with a failure to disambiguate

### Version of Marlin Firmware

The latest -bugfix zip, 4 Aug 2024

### Printer model

Purpose built binder jetting machine for b-jetting dot com

### Electronics

Octopus Pro V1.1

### LCD/Controller

None

### Other add-ons

I have all 9 axis enabled as linear axis, no extruders

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_Configs.zip](https://github.com/user-attachments/files/16496082/Marlin_Configs.zip)
[pins_files.zip](https://github.com/user-attachments/files/16496116/pins_files.zip)

pins_files include the mods necessary for 9 linear axis on a BTT Octopus Pro V1.1

As for the slicing and host software. These are proprietary and custom built for binder jetting, not FDM. I wrote them myself.

DerAndere1, 2024-08-08:
Well spottet. Thanks for the Report. @thebjtfellow Do you want to create a pull request targeting bugfix-2.1.x to fix this? Or do you prefer me doing it? Kind regards 
DerAndere

thebjtfellow, 2024-08-08:
Hi DerAndere! First, let me thank you for writing the PipetteBot extension to Marlin! Making a binder jetting machine on this platform wouldn't have been possible without it, honestly. It's a long story, but thank you!

I would prefer if you submit the pull request please. I'm still trying to learn github and how this all works when it's working right.

Again, thank you!!

commit 9a2c7f92 (2024-08-12): 🩹 Fix W axis blocking move

Fixes #27332
commit c7929217 (2024-08-12): 🩹 Fix W axis blocking move

Fixes #27332
commit 37fb26b5 (2024-08-12): 🩹 Fix W axis blocking move

Fixes #27332
closed 2024-08-12 by thinkyhead