MarlinFirmware/Marlin #27333: [BUG] Marlin Crashes after any Stepper move with TMC2226 drivers

KevWal, 2024-08-06:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I have had Marlin 2.1.2.4 working on my printer fine with A4988 drivers , but just replaced the stepper drivers with BTT TMC2226 v1.0's, updated the configuration, uploaded and now any stepper move results in a crash of the firmware.

When I say 'crash'. it becomes un responsive on either serial console, unresponsive on the LCD, the (blue in my case) LED1 connected to D13 flashes rapidly (it doesnt normally flash at all other than twice on boot up), and whilst pressing the reset button in stops the blue LED flashing, on releaseing reset the blue LED starts flashing again and the board does not reset.  Only way to get back to Marlin is a full power cycle of the PSU.

I can communicate with the TMC2226's just fine, and get debuging out of them untill the moment before the crash:

Point of crash:
```
Recv: X:20/17/-|GI|	Y:20/17/-|GI|	Z:20/17/-|GI|	E:20/17/-|GI|
Send: G28 Y
Recv: X:20/17/-|GI|	Y:20/17/-|GI|	Z:32/17/-|G|	E:20/17/-|GI|
[Blue LED starts flashing]
```

Just before the LED1 starts flashing I sometimes hear a small click - like a stepper is just starting to move.

Other functions - bed heating, hot end heating, fan control all work fine if tested before I try and move a stepper.

TMC Debug messages:
```
Send: M122 P10
[...]
Recv: axis:pwm_scale/curr_scale/mech_load|flags|warncount
Recv: 		X	Y	Z	E
Recv: Address		0	1	2	3
Recv: Enabled		false	false	false	false
Recv: Set current	800	800	800	800
Recv: RMS current	760	760	760	760
Recv: MAX current	1072	1072	1072	1072
Recv: Run current	17/31	17/31	17/31	17/31
Recv: Hold current	17/31	17/31	17/31	17/31
Recv: CS actual	17/31	17/31	17/31	17/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false
Recv: msteps		64	64	64	64
Recv: interp		false	false	false	false
Recv: tstep		max	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum	20	20	20	20
Recv: pwm scale auto	0	0	0	0
Recv: pwm offset auto	36	36	36	36
Recv: pwm grad auto	14	14	14	14
Recv: off time	0	0	0	0
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv:  -end		-1	-1	-1	-1
Recv:  -start		1	1	1	1
Recv: Stallguard thrs	0	0	0	0
Recv: uStep count	2	2	2	2
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0x80:11:00:00
Recv: 		Y	0x80:11:00:00
Recv: 		Z	0x80:11:00:00
Recv: 		E	0x80:11:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK
Recv: ok P15 B3

```

Startup messages:
```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyAMA0, baudrate 250000
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6f12f6e8, open=True>(port='/dev/ttyAMA0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: o\x00k N0 P15 B3
WARNThe received line contains at least one null byte character at position 1, this hints at some data corruption going on
Recv: //action:notification 3D Printer Ready.
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
Recv: ok N0 P15 B3
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok N0 P15 B3
Send: N0 M110 N0*125
Recv: ok N0 P15 B3
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.1.2.4 (Aug  6 2024 09:31:56) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:1
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:1
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:MULTI_VOLUME:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:1
Recv: Cap:LONG_FILENAME:0
Recv: Cap:LFN_WRITE:0
Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0
Recv: Cap:EXTENDED_M20:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:1
Recv: area:{full:{min:{x:0.00,y:0.00,z:0.00},max:{x:190.00,y:210.00,z:185.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:190.00,y:210.00,z:185.00}}}
Recv: ok N1 P15 B3
Recv: //action:notification 3D Printer Ready.
Send: M21
Recv: echo:SD card ok
Recv: ok P15 B3
Send: M503
Recv: echo:; Linear Units:
Recv: echo:  G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D2.85
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z4040.00 E795.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z2.50 E25.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P3000.00 R3000.00 T3000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.01
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00 ; Leveling OFF
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H210.00 B40.00 F0
Recv: echo:  M145 S1 H240.00 B100.00 F0
Recv: echo:; Hotend PID:
Recv: echo:  M301 P13.26 I0.54 D80.90
Recv: echo:; Controller Fan:
Recv: echo:  M710 S127 I0 A1 D60 ; (49% 0%)
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-25.00 Y0.00 Z-1.55 ; (mm)
Recv: echo:; Stepper driver current:
Recv: echo:  M906 X800 Y800 Z800
Recv: echo:  M906 T0 E800
Recv: echo:; Driver stepping mode:
Recv: echo:; Backlash compensation:
Recv: echo:  M425 F1.00 X0.00 Y0.00 Z0.00
Recv: ok P15 B3
Send: M78
Recv: echo:Unknown command: "M78"
Recv: ok P15 B3
Send: M154 S5
Recv: ok P15 B3
Send: M155 S2
Recv: ok P15 B3
Send: M876 P1
Recv: ok P15 B3
[...]
Recv: ok P15 B3
Send: M20
Recv: Begin file list
Recv: End file list
Recv: ok P15 B3
Send: M20
Recv: Begin file list
Recv: End file list
Recv: ok P15 B3
```

### Bug Timeline

Noticed 06/08/2024

### Expected behavior

I expect the stepper to move

### Actual behavior

The firmware appears to crash, although LED1 flashes continually.

### Steps to Reproduce

Issue any GCODE that tries to home any of the steppers

### Version of Marlin Firmware

2.1.2.4

### Printer model

GC I3 - https://web.archive.org/web/20160411113355/http://www.emakershop.com/browse/listing?l=964

### Electronics

RAMPS 1.4

### LCD/Controller

Panelolu2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/16512304/Config.zip)

ellensp, 2024-08-06:
disable #define SOFTWARE_DRIVER_ENABLE

Your  steppers have an enable IO pin

KevWal, 2024-08-06:
Between my last working config, and this crashing config I did also enable  `#define ENDSTOP_INTERRUPTS_FEATURE`.

I have now disabled that and removed my `#define SW_CAPABLE_PLATFORM false` in TMCStepper.h that allowed that to work but still no change.  Still seems to crash and leave me with the rapid flashing LED1 and a non working reset button.

KevWal, 2024-08-06:
> disable #define SOFTWARE_DRIVER_ENABLE
> 
> Your steppers have an enable IO pin

Thanks @ellensp , I have tried that, as well as enabling #define WATCHDOG_RESET_MANUAL and now I get a soft "Error:Watchdog timeout. Reset required."  rather than a crash I can not reset from.

[config2.zip](https://github.com/user-attachments/files/16514285/config2.zip)

```
19:15:47.279 >  External Reset
19:15:47.279 > Marlin 2.1.2.4
19:15:47.279 > echo: Last Updated: 2024-06-16 | Author: (Kevin Walton, Aug 2024, Based on GC original with Y and TMC upgrade)
19:15:47.284 > echo: Compiled: Aug  6 2024
19:15:47.284 > echo: Free Memory: 356  PlannerBufferBytes: 1200
19:15:47.464 > //action:notification 3D Printer Ready.
19:15:48.401 > echo:V88 stored settings retrieved (679 bytes; crc 41630)
19:15:48.407 > //action:notification Stored settings retrieved
19:15:48.407 > echo:; Linear Units:
19:15:48.407 > echo:  G21 ; (mm)
19:15:48.407 > echo:; Temperature Units:
19:15:48.407 > echo:  M149 C ; Units in Celsius
19:15:48.407 > echo:; Filament settings (Disabled):
19:15:48.407 > echo:  M200 S0 D2.85
19:15:48.407 > echo:; Steps per unit:
19:15:48.407 > echo:  M92 X80.00 Y80.00 Z4040.00 E795.00
19:15:48.407 > echo:; Max feedrates (units/s):
19:15:48.407 > echo:  M203 X300.00 Y300.00 Z2.50 E25.00
19:15:48.418 > echo:; Max Acceleration (units/s2):
19:15:48.418 > echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
19:15:48.423 > echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
19:15:48.423 > echo:  M204 P3000.00 R3000.00 T3000.00
19:15:48.434 > echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
19:15:48.434 > echo:  M205 B20000.00 S0.00 T0.00 J0.01
19:15:48.434 > echo:; Home offset:
19:15:48.434 > echo:  M206 X0.00 Y0.00 Z0.00
19:15:48.434 > echo:; Auto Bed Leveling:
19:15:48.434 > echo:  M420 S0 Z10.00 ; Leveling OFF
19:15:48.434 > echo:; Material heatup parameters:
19:15:48.442 > echo:  M145 S0 H210.00 B40.00 F0
19:15:48.442 > echo:  M145 S1 H240.00 B100.00 F0
19:15:48.442 > echo:; Hotend PID:
19:15:48.442 > echo:  M301 P13.26 I0.54 D80.90
19:15:48.442 > echo:; Controller Fan:
19:15:48.442 > echo:  M710 S127 I0 A1 D60 ; (49% 0%)
19:15:48.442 > echo:; Z-Probe Offset:
19:15:48.446 > echo:  M851 X-25.00 Y0.00 Z-1.55 ; (mm)
19:15:48.446 > echo:; Stepper driver current:
19:15:48.446 > echo:  M906 X800 Y800 Z800
19:15:48.446 > echo:  M906 T0 E800
19:15:48.456 > echo:; Driver stepping mode:
19:15:48.456 > echo:; Backlash compensation:
19:15:48.456 > echo:  M425 F1.00 X0.00 Y0.00 Z0.00
19:15:52.190 > //action:prompt_end
19:15:52.394 > Testing X connection... OK
19:15:52.403 > Testing Y connection... OK
19:15:52.407 > Testing Z connection... OK
19:15:52.411 > Testing E connection... OK
19:15:52.927 > echo:SD card ok

19:16:18.118 > //action:notification 3D Printer Ready.
G28 Y
19:16:27.402 > Error:Watchdog timeout. Reset required.
19:16:28.285 > Po
```

closed 2024-11-20 by github-actions[bot]