MarlinFirmware/Marlin #27336: dot printing when powerloss is ON

Enzo-Enz, 2024-08-07:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

HI guys
when poweloss recovery is ON,while the printer starts to print curve side,It prints like single dot as ypu see in the video
but when I turn off poweloss recovery It prints smoothly

https://github.com/user-attachments/assets/f8ca4108-e25b-4e18-946b-54f802789082

### Bug Timeline

new

### Expected behavior

print smoothly

### Version of Marlin Firmware

2.1.2.4

### Printer model

diy printer

### Electronics

mks tinybee

### LCD/Controller

128x64

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/16533537/config.zip)

ellensp, 2024-08-08:
You have the worst type of power loss recovery, where it writes to the sdcard lots as it has now power loss detection hardware.

Check your sdcard, reformat it, and make sure its fast

Enzo-Enz, 2024-08-10:
> You have the worst type of power loss recovery, where it writes to the sdcard lots as it has no power loss detection hardware.
> 
> Check your sdcard, reformat it, and make sure its fast

thank you bro for your reply 
I did that,It's better than before but the problem still exist in higher speed

thinkyhead, 2024-08-12:
This is a problem that exists for the SD-card-based power-loss recovery when there is no power loss sensor. Every write to the SD card causes some small delay in the printing, and this is especially noticeable with vase mode printing. We could add a setting that makes power-loss recovery only start after X layers, but the issue you describe is likely to affect any layer with a similar shape, so it's not a solution.

We could also make a version that writes to the EEPROM, probably without these delay glitches, but that leads to the problem of potentially wearing out your EEPROM, or even your MCU's own flash storage, which would not be good.

The only solution for you right now will be to add a power-loss detection circuit to your printer, or don't enable power-loss recovery until later in the print, or don't use this rather crude power-loss recovery feature at all.

closed 2024-12-05 by github-actions[bot]