MarlinFirmware/Marlin #27339: [BUG] Can't compile due toolchain-gccarmnoneeabi errors on LPC1768

zonalimitatore, 2024-08-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling the last bugfix, compiler return:
```
Archiving .pio\build\LPC1768\libFrameworkArduino.a
Linking .pio\build\LPC1768\firmware.elf
lf section `.bss' will not fit in region `RAM'
c:/users/bax/.platformio/packages/toolchain-gccarmnoneeabi@1.100301.220327/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1768/firmware.elf section `.AHBSRAM1' will not fit in region `ETH_RAM'
c:/users/bax/.platformio/packages/toolchain-gccarmnoneeabi@1.100301.220327/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
c:/users/bax/.platformio/packages/toolchain-gccarmnoneeabi@1.100301.220327/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 680 bytes
c:/users/bax/.platformio/packages/toolchain-gccarmnoneeabi@1.100301.220327/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: region `ETH_RAM' overflowed by 55536 bytes
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1768\firmware.elf] Error 1
```

### Bug Timeline

I can compile the last stable, not this bugfix

### Expected behavior

Compile 

### Actual behavior

Errors at the end of the compile work

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Aug  5 2024 23:22:11) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Stampante KINEMATICS:Cartesian EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

### Electronics

SKR 1.4 non turbo

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.txt](https://github.com/user-attachments/files/16554412/Configuration_adv.txt)
[Configuration.txt](https://github.com/user-attachments/files/16554415/Configuration.txt)

thisiskeithb, 2024-08-09:
> lf section `.bss' will not fit in region `RAM'

Disable some features since you've exceeded your boards' RAM limit.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2024-08-09 by thisiskeithb

zonalimitatore, 2024-08-09:
I've enable the same things of the stable. Remain 50% of ram. 
I can't compile the bugfix. Isn't Ram limit issue

thisiskeithb, 2024-08-09:
> I've enable the same things of the stable. Remain 50% of ram. I can't compile the bugfix. Isn't Ram limit issue

Please attach a .zip with your `2.1.2.4` configs.

zonalimitatore, 2024-08-09:
[Marlin.zip](https://github.com/user-attachments/files/16554594/Marlin.zip)

zonalimitatore, 2024-08-09:
If i disable 
#define INPUT_SHAPING_X
#define INPUT_SHAPING_Y

It compiles normally
They take so much ram?

thisiskeithb, 2024-08-09:
> They take so much ram?

Yes, it uses a lot of ram and fitting IS on LPC can particularly tricky.

zonalimitatore, 2024-08-09:
Got it, thanks