MarlinFirmware/Marlin #27341: [BUG] Dual Z and EDGE_STEPPING makes first Z rattle and miss steps

Sanguium, 2024-08-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When activating a second Z driver and EDGE_STEPPING, the first Z motor grinds and loses steps, while the second works fine.
All other axis work fine, problem goes away when disabling the Z2 driver OR disabling EDGE_STEPPING.

This is the M122 report after moving Z 100m up with and without EDGE_STEPPING enabled:
EDGE_STEPPING OFF:
[image]
EDGE_STEPPING ON:
[image]

Using SKR v1.3 with TMC2009 drivers

### Bug Timeline

Tested wrong in 2.1.2.4 and current bugfix-2.1.x

### Expected behavior

Dual Z drivers work fine with the recomended EDGE_STEPPING for trinamic drivers

### Actual behavior

First Z motor rattles and loses steps

### Steps to Reproduce

1. Set `Z_DRIVER_TYPE  TMC2209` and `Z2_DRIVER_TYPE TMC2209`
2. Set `EDGE_STEPPING`
3. Move Z axis

### Version of Marlin Firmware

Commit 0ec1a543

### Printer model

Custom cartesian

### Electronics

SKR v1.3 baord with TMC2209 drivers

### LCD/Controller

BIGTREE_TFT24_V1_1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/16566144/Config.zip)

anomalchik, 2024-08-13:
Same issue with mks sgen l v1
I use tmc2226 and they also use the driver type TMC2209

anomalchik, 2024-08-13:
If you turn off the first motor but do not remove the driver, the second motor remains in the brake mode or something like that. Also, the second motor works properly if you remove the Z0 driver.

ellensp, 2024-11-04:
@tombrazier TOGGLE was recently changed https://github.com/MarlinFirmware/Marlin/pull/27149 ...

tombrazier, 2024-11-04:
Okay, @moriamoria, @anomalchik and @Sanguium here is something to try:

Change the file `Marlin/src/HAL/LPC1768/fastio.h` as follows:

Remove `#define _TOGGLE(IO)           LPC176x::gpio_toggle(IO)`

and replace it with `#define _TOGGLE(IO)           _WRITE(IO, !READ(IO))`.

Does this resolve the problem?

I have a bunch of mainboards but no LPC1768 to test on.

moriamoria, 2024-11-04:
Hello,

I have kill my SKR 1.3 motherboard, the X- Y- pullup end switch seems dead !

I have order a new SKR 1.4 turbo, and i will do the test when I receive it.

Thanks for your support

Regards

Sanguium, 2024-11-04:
> Okay, @moriamoria, @anomalchik and @Sanguium here is something to try:
>
> Change the file `Marlin/src/HAL/LPC1768/fastio.h` as follows:
>
> Remove `#define _TOGGLE(IO) LPC176x::gpio_toggle(IO)`
>
> and [...]

Tested with latest bugfix, this change does fix the problem.
Can I keep this until the fix is merged or does it have some other caveats?

Thanks.

tombrazier, 2024-11-04:
@mh-dm please see above and also #27489. It looks like #27149 has caused a problem with dual Z steppers. My guess is that two calls to `LPC176x::gpio_toggle()` in quick succession can cause some kind of race condition. Any thoughts?

tombrazier, 2024-11-04:
I think I have found the race condition. Function `LPC176x::gpio_toggle(IO)` is defined in @p3p's [PlatformIO framework for the LPC1768](https://github.com/p3p/pio-framework-arduino-lpc176x) in file `master/system/lpc176x/pin_control.h`.

The function is
```
  [[gnu::always_inline]] inline void toggle() {
    gpio_reg().reg_control ^= gpio_mask();
  }
```

Reading chapter 9 of the LPC1768 user manual (a copy can be found [here](https://www.keil.com/dd/docs/datashts/philips/lpc17xx_um.pdf)) this function reads `FIOPIN`, modifies it and sets it to the new value. However a read from `FIOPIN` returns the current _pin state_, not the current _port output register_. To get the latter you have to read `FIOSET`.

As a consequence, when two pin toggles occur in quick succession the second read of the pin state can happen before the pin has actually changed after the first toggle operation.

I think the function should be modified to:
```
  [[gnu::always_inline]] inline void toggle() {
    gpio_reg().reg_control = gpio_reg().reg_set ^ gpio_mask();
  }
```

If there is anyone who is able to find their platformio directory (on Linux it is `~/.platformio`) and change the file `packages/framework-arduino-lpc176x/system/lpc176x/pin_control.h` as I suggest I would be interested to know the result.

Also any comment by @p3p would be appreciated.

tombrazier, 2024-11-04:
> Can I keep this until the fix is merged or does it have some other caveats?

@Sanguium Yes you can keep it until a fix is merged. The difference is slightly less clean step timing, but that's way better than steps not actually happening.

PS If you're able to test the change to `LPC176x::gpio_toggle()` I would appreciate it.

Sanguium, 2024-11-04:
> PS If you're able to test the change to `LPC176x::gpio_toggle()` I would appreciate it.

This also works.

tombrazier, 2024-11-04:
https://github.com/p3p/pio-framework-arduino-lpc176x/pull/55

tombrazier, 2024-11-04:
@mh-dm Do you have an LPC mainboard you could do some testing on? I'd like to see this tested fairly widely considering it affects the underlying framework.

anomalchik, 2024-11-05:
I'll try to check it today.

anomalchik, 2024-11-05:
> ```
>   [[gnu::always_inline]] inline void toggle() {
>     gpio_reg().reg_control = gpio_reg().reg_set ^ gpio_mask();
>   }
> ```
@tombrazier
works fine for me. Thanks. My board is MKS_SGEN_L LPC1768.

tombrazier, 2024-11-05:
@anomalchik Thanks.

mh-dm, 2024-11-10:
> @mh-dm Do you have an LPC mainboard you could do some testing on? I'd like to see this tested fairly widely considering it affects the underlying framework.

Sure. Tested in the same scenario as in #27149 and can confirm that the proposed `toggle()` version works including maintaining the slightly more consistent timings/speed benefit over the original `_WRITE(IO, !READ(IO))`.

tombrazier, 2024-11-10:
Thanks Mihai.

thinkyhead, 2024-11-19:
There will be a new deployment of @p3p pio-framework-arduino-lpc176x very soon, so hold tight! We'll be sure to get this back-ported to all LTS branches as soon as possible too.

closed 2024-11-30 by thinkyhead