MarlinFirmware/Marlin #27343: [BUG] FT_MOTION - no censorless sensitivity could be tuned

vovodroid, 2024-08-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With FT_MOTION enabled (no both old and new IS) it's impossible to find good value for censorless sensitivity. For instance there is false triggering with 88, but no bump detection with 87.

BTT SKR 3, TMC2209, homing current 100mA

### Version of Marlin Firmware

bugfix-2.1.x 2024-08-07

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16569542/Marlin.zip)

thisiskeithb, 2024-08-10:
> homing current 100mA

That's pretty low. Try bumping that up in 100mA increments. I typically run 1/2 `X/Y_CURRENT` for `X/Y_CURRENT_HOME` with sensorless homing.

You can also try increasing `HOMING_FEEDRATE_MM_M`. I have to home slowly on some machines, while others have to home quickly for sensorless to work.

vovodroid, 2024-08-10:
Question is why FTM affects these parameters? Isn't it internal driver function?

thisiskeithb, 2024-08-10:
> Question is why FTM affects these parameters? Isn't it internal driver function?

It's a whole new motion system & still quite buggy / experimental.

narno2202, 2024-08-10:
I have the same problem when I set sensorless homing for the fisrt time in the standard motion system. I have to decrease the X and Y motor current to get reliable homing. For the X axis nominal current is 800 and 400 for homing, for Y axis nominal current is 1000 and 800 for homing.
With `FT_MOTION` I need to change the stallguard sensiitivty : X senssitivity 55 no change,  Y sensitivity 45 instead of 55.

vovodroid, 2024-08-11:
I played with current in range 200-400, but didn't succeeded to get it working.

I saw proposition to disable FTM during homing, why not to do it?

narno2202, 2024-08-11:
I forgot to mention that with my settings, sensorless homing works both with standard motion and FTM.  What kind of failure do you have? : no motion or grinding on stop.

vovodroid, 2024-08-11:
> grinding on stop.

This one or false triggering.

thinkyhead, 2024-08-11:
> I saw proposition to disable FTM during homing, why not to do it?

If homing doesn't work with FT Motion then this is a broken aspect of FT Motion that we want to fix, especially if FTM is going to be the only motion system enabled for some builds. Eventually we'll get it working reliably.

I'm not sure if Ulendo has sensorless homing setups they can use for testing and debugging, but I sure do, and as I get into deeper testing along with Ulendo over the coming days I'm sure we'll find and fix the cause of the unreliability.

narno2202, 2024-08-12:
I have found sensorless homing difficult to tune with  standard motion. Once tuned, It works fine also with FTM with minor changes but seems to be relatively hardware dependant.
@thinkyhead, Ulendo tested FTM on an Ender 3 style printer (as mentioned in their scientific papers). In previous talks with @ulendoalex, I've understood that FTM was not tested with sensorless homing and diffrent kind of probe. This was fixed after merging into Marlin.

vovodroid, 2024-08-12:
Could sensorless homing issue related to sound? Actually strange sound is caused by current waves, probably affecting bump detection.

narno2202, 2024-08-12:
I don't think so.
I have a look at your `Configuration.h` file. Homing speed is 20mm/s  (X and Y axis). Sensorless homing also relies on speed, mine is Marlin default (50mm/s). The 3 parameters are speed (not too low not to high), motor current and stallguard sensitivity. You need to tweak them to find the good combination but in my experience It's a little bit tricky.

vovodroid, 2024-08-12:
> Homing speed is 20mm/s (X and Y axis).

That's Klipper recommendation https://www.klipper3d.org/TMC_Drivers.html#choose-homing-speed

```
A good starting point for the homing speed is for the stepper motor to make a full rotation every two seconds.
For many axes this will be the rotation_distance divided by two. For example:

[stepper_x]
rotation_distance: 40
homing_speed: 20
```

> You need to tweak them to find the good combination but in my experience It's a little bit tricky.

I had no problem with tuning standard motion. Speed and current affect sensitivity, but there is always range 50 or more where detection works.

vovodroid, 2024-08-29:
Ok, I've tested with increased speed - 40-50 mm/s. Now I succeeded to to home X/Y, though sensitivity range is still more narrow than for "classic" motion.

thinkyhead, 2024-12-04:
Excellent! We'll add a note to the homing speed and maybe even provide a sensible default based on this.
```
A good starting point for the homing speed is for the stepper motor to make a full rotation every two seconds.
For many axes this will be the rotation_distance divided by two. For example:
```

closed 2024-12-04 by thinkyhead