MarlinFirmware/Marlin #27348: [BUG] Marlin 2.1.2.4 causes layer shift one every layer only on COREXY machine.

YHamud, 2024-08-11:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Everything compiles fine and when used on a standard cartesian machine it works flawlesly. I took the same files and enabled corexy and made some other changes refering to the machine size, leveling sensor, etc. Compiled fine, everything works, but it causes layer shifts on each layer. It's like the calculations for the corexy motions are wrong.

I've been using the bugfix (2.1.3) from the begining of the year without major issues, but wanted to try the latest oficial release.

### Bug Timeline

New bug

### Expected behavior

I expected it to not produce layer shifts.

### Actual behavior

It produces layer shifts in diagonal on each layer.

### Version of Marlin Firmware

2.1.2.4

### Printer model

TwoTrees SP5V3

### Electronics

MKS Robin Nano V3.1

### LCD/Controller

TS35

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[CONFIGURATIONS_MARLIN.zip](https://github.com/user-attachments/files/16548705/CONFIGURATIONS_MARLIN.zip)

thisiskeithb, 2024-08-11:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

YHamud, 2024-08-16:
> > I've been using the bugfix (2.1.3) from the begining of the year without major issues, but wanted to try the latest oficial release.
> 
> `bugfix-2.1.x` is more than 1.5 years ahead of any tagged release at this point since releases are no longer snapshots of the bugfix branch. This includes more hardware support & many fixes.
> 
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Thanks for your reply.

It works well with bugfix (configuration file version 2.1.3), that's what I've been using. But then I have another problem, everytime I try Z babystepping during the first layer the machine behaves funny. Sometimes the nozzle goes back on it's trajectory very slowly then rushes to the last position, other times the bed lowers a lot then comes back up.

ucirello, 2024-08-22:
I'm seeing something similar:
[image]
[Archive.zip](https://github.com/user-attachments/files/16705390/Archive.zip)

ucirello, 2024-08-22:
I can confirm that with 12a2e5ae1b6494722a7f541845139ac19424a64a (current HEAD of [bugfix-2.1](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.1.x/) it works correctly.

houseofbugs, 2024-08-28:
Im in the process of upgrading our UF2 code base to 2.1.2.4. Tested on 2 CoreXY machines (One of mine and one of my remote guys) and neither are able to reproduce this issue over dozens of prints. I'd suspect driver current and/or sense resistor settings being the culprit here.

ucirello, 2024-08-28:
@houseofbugs 
> I'd suspect driver current and/or sense resistor settings being the culprit here.

Possibly? In terms of Configuration.h and Configuration_adv.h both RSense and current were the same before and after. I wonder if the internals reading these values may have changed somewhere.

In any case, there was a lot of noise in the diff when I transposed the configuration from `2.1.2.4` to `bugfix-2.1.x`, so it's not impossible that I may have unintentionally change some parameter.

ucirello, 2024-11-27:
bugfix-2.1.x works!

saintofinternet, 2025-02-07:
If you check this version I was having a great first print after the firmware flash but when I repeated the same print I get the layer shift. 

What could be the reason as there is no mechanical issue which I checked again and again. 

[image]

[image]

[image]

tocsa120ls, 2025-04-24:
I had this issue after upgrading my Ender5 with the Endorphin mod. The software is all right, but you need to crank the Vref voltage of the X stepper up. From Creality the board comes with every single Vref set to 1.1V which is safe for the default config. But a CoreXY mod adds a lot of belt and friction to the X servo. I have mine set on the X axis stepper controller to 1.3V. This cured the missed steps completely and the stepper is certainly warmer but not untouchable.