MarlinFirmware/Marlin #27354: [BUG] Latest bugfix won't home during G33 delta calibration

rq3, 2024-08-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The bugfix as of August 16, 2024 fails to home during a G33 delta calibration. After the last probe point, when it should gracefully home and then start the next iteration, it randomly moves at high speed in any axis.

### Bug Timeline

new

### Expected behavior

I expect it to complete G33.

### Actual behavior

It randomly moves at high speed on a random axis when it should be homing after a G33 iteration.

### Steps to Reproduce

1) Compile and install latest bugfix
2) Start G33 delta calibration
3) After probing, observe effector wildly head in random directions at high speed, when it should just quietly home.

### Version of Marlin Firmware

latest bugfix (Aug. 16, 2024)

### Printer model

Anycubic Predator

### Electronics

Stock Trigorilla Pro (STM32F103ZETA)

### LCD/Controller

Anycubic TFT 3.5

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/16641941/config.zip)

thinkyhead, 2024-08-17:
Would you be able to test some `bugfix-2.1.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

renzaaa, 2024-08-18:
G33 delta calibration works on 2024-07-27 bugfix.

rq3, 2024-08-18:
Fails in bugfix 89b0143b09 from Aug.11.

thisiskeithb, 2024-08-18:
Thanks for tracking it down!

> Fails in bugfix [89b0143](https://github.com/MarlinFirmware/Marlin/commit/89b0143b09aeb617d17cc029d3bbb363dd1c857b) from Aug.11.

It actually worked in our favor since you found the exact commit, but there must have been a GitHub glitch since all the commits from https://github.com/MarlinFirmware/Marlin/pull/27346 were merged into`bugfix-2.1.x` individually instead of the usual "Squash and merge"/single commit (with the referenced PR # in the title) that's usually done.

rq3, 2024-08-18:
> Thanks for tracking it down!
>
>
> It actually worked in our favor since you found the exact commit, but there must have been a GitHub glitch since all the commits from #27346 were merged [...]

My apologies, @thisiskeithb. My last post was badly worded, as it implies that the failure is in 89b0143. I should have said that it "also fails in 89b0143". I have not checked prior to that, as I am having trouble locating the full commits. However, we do know that it was good on July 27, and bad on August 11.

thisiskeithb, 2024-08-21:
> I am having trouble locating the full commits.

You can always find all commits by clicking on the "Commits" section from the main repo page:

[image]

This will lead you to the [full commit history](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.1.x/) where you can browse the code from any point in time:

[image]

---

> I have not checked prior to that

So close! 89b0143b09aeb617d17cc029d3bbb363dd1c857b is likely the source of issues for you since it causes *really* fast X/Y travel move to `Z_SAFE_HOMING_X_POINT`/`Z_SAFE_HOMING_Y_POINT` during `G28` for non-Delta configs.

Resetting to right before that commit (i.e. 0790a9d5df30f4a6a5990482d78f0b5c85ec9508) will likely fix the reported issue, so give it a try: [.ZIP link](https://github.com/MarlinFirmware/Marlin/archive/0790a9d5df30f4a6a5990482d78f0b5c85ec9508.zip)

Here's a [.ZIP link](https://github.com/MarlinFirmware/Configurations/archive/dc8b710af83581359ef8cf941312e61604f25526.zip) of compatible Configurations if you don't want to back port your config. Note: You'll need to remove the two "`#error`" lines at the top of `Configuration.h` if you use these configs since they originate from the `import-2.1.x` branch.

rq3, 2024-08-21:
Bingo. 0790a9d works. The next commit, 89b0143, does not. Just as you suspected, @thisiskeithb.

thisiskeithb, 2024-08-24:
@rq3: Can you try thinkyhead’s suggestion? https://github.com/MarlinFirmware/Marlin/pull/27349#issuecomment-2308070643

rq3, 2024-08-24:
> @rq3: Can you try thinkyhead’s suggestion? [#27349 (comment)](https://github.com/MarlinFirmware/Marlin/pull/27349#issuecomment-2308070643)
> 
> Edit: thinkyhead snuck [e2d8b2f](https://github.com/MarlinFirmware/Marlin/commit/e2d8b2fc26b74af150eace5eb5e965223dcc6e2d) in, so grab the latest bugfix and give it a test. This fixes the _super fast_ x/y homing move on my i3 machines.

I'm on it. Will report back soon.

rq3, 2024-08-24:
Latest bugfix works perfectly! Thanks guys! Issue closed.

[image]

https://github.com/user-attachments/assets/18ec9813-c652-4d1e-98d9-d7e2736e80f0

closed 2024-08-24 by rq3