MarlinFirmware/Marlin #27358: [BUG] Sensorless homing only works when homing manually with display menu and not from SD G28 G-code

GITRDUN75, 2024-08-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using TMC2209 drivers on an Octopus motherboard with a full graphics smart controller. Machine is custom made coreXY. Have tried the latest bugfix version and also a couple older versions of Marlin. New machine build so this isnt something that was working and now is not.
 Sensorless homing works as expected if i click HOME ALL on the display menu. 
 Does not work when a G28, G28 X or G28 Y is issued from G code on a SD card. Instead the machine will begin homing X and or Y and will vibrate the motors when they physically run up against the rails but will not home.

### Bug Timeline

not sure. New machine build

### Expected behavior

When the machine is given a G28 command from the G-code file it is running it should home XY then Z axes.

### Actual behavior

When the machine is given a G28 command via G-code on the SD card reader the machine attempts to home X and Y axis but does not sense the stall and does not complete the homing sequence. However if the machine is homed using the HOME ALL command from the UI display menu it will home as it should. Also once homed correctly it will not home again via the UI unless the power is cycled off / on.

### Steps to Reproduce

Compile and load latest Marlin 2.1 or 2.1 bugfix.
Home machine using HOME ALL from the UI. Machine will home normally.
Run a G-code program which contains a G28 from the onboard SD card. Machine will attempt to home but will not sense the motors stalling and will continue turning motors.

### Version of Marlin Firmware

Marlin-Bugfix-2.1X ,    Marlin-2.0.9.7

### Printer model

CoreXY custom

### Electronics

BTT Octopus

### LCD/Controller

reprap discount full graphics smart controller

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/16651446/Configuration.zip)
[Configuration_adv.zip](https://github.com/user-attachments/files/16651448/Configuration_adv.zip)

GITRDUN75, 2024-08-22:
I have other printers using this same electronics setup only are not coreXY kinematics. They do not have this problem. I have tried using the same Marlin version as some of my other printers which are not coreXY and the problem is there only for coreXY.

Persei1980, 2024-08-28:
У меня была така же проблема, отредактируйте Configuration_adv.h  начиная со строки 1520 #if ENABLED(SDSUPPORT).

GITRDUN75, 2024-08-28:
> У меня была така же проблема, отредактируйте Configuration_adv.h начиная со строки 1520 #if ENABLED(SDSUPPORT).

Translated (
I had the same problem, edit Configuration_adv.h starting from line 1520 #if ENABLED(SDSUPPORT). )
 
 What SD setting was causing your problem? 
  I will toy around with the SD settings this evening.

GITRDUN75, 2024-09-21:
Still havent solved this but have figured out that it definitely has something to do with the SD card being active. If #define SD_IGNORE_AT_STARTUP is uncommented in config.adv file the machine will home fine using the LCD menu. But as soon as you load the SD card and it is activated the machine will no longer home either using the LCD menu or from a G-code file.

GITRDUN75, 2024-11-21:
Think i have this issue resolved. I had the jumpers for the SPI / UART installed as SPI and i am using UART drivers. Somewhere along the way i got mixed up and had them set incorrectly. I assume that was interfering with the LCD connections somehow.