MarlinFirmware/Marlin #27372: [BUG] (bug summary)

Minims, 2024-08-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Since https://github.com/MarlinFirmware/Marlin/commit/94e9f26544d77d22c67a8973a4b480411ecfe7d6

I have this error : 

```
Compiling .pio/build/LPC1769/src/src/module/settings.cpp.o
In file included from Marlin/src/gcode/control/../../inc/MarlinConfigPre.h:43,
                 from Marlin/src/gcode/control/../../inc/MarlinConfig.h:28,
                 from Marlin/src/gcode/control/../gcode.h:342,
                 from Marlin/src/gcode/control/M80_M81.cpp:23:
Marlin/src/gcode/control/M80_M81.cpp: In static member function 'static void GcodeSuite::M81()':
Marlin/src/gcode/control/M80_M81.cpp:87:30: error: 'recovery' was not declared in this scope; did you mean 'remove'?
   87 |   TERN_(POWER_LOSS_RECOVERY, recovery.purge()); // Clear PLR on intentional shutdown
      |                              ^~~~~~~~
Marlin/src/gcode/control/../../inc/../core/macros.h:620:26: note: in definition of macro 'THIRD'
  620 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin/src/gcode/control/../../inc/../core/macros.h:208:29: note: in expansion of macro '___TERN'
  208 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin/src/gcode/control/../../inc/../core/macros.h:207:29: note: in expansion of macro '__TERN'
  207 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin/src/gcode/control/../../inc/../core/macros.h:206:29: note: in expansion of macro '_TERN'
  206 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
      |                             ^~~~~
Marlin/src/gcode/control/M80_M81.cpp:87:3: note: in expansion of macro 'TERN_'
   87 |   TERN_(POWER_LOSS_RECOVERY, recovery.purge()); // Clear PLR on intentional shutdown
      |   ^~~~~
Marlin/src/inc/Warnings.cpp:238:6: warning: #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)" [-Wcpp]
  238 |     #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"
      |      ^~~~~~~
*** [.pio/build/LPC1769/src/src/gcode/control/M80_M81.cpp.o] Error 1
======================================================================== [FAILED] Took 3.38 seconds ========================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:03.377
=================================================================== 1 failed, 0 succeeded in 00:00:03.377 ===================================================================
```

### Bug Timeline

New bug, starts with https://github.com/MarlinFirmware/Marlin/commit/94e9f26544d77d22c67a8973a4b480411ecfe7d6

### Expected behavior

Build does not fail

### Actual behavior

Build is failling

### Steps to Reproduce

1. Build marlin with this option :

```
 #define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT        true // Power-Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)
    //#define PLR_BED_THRESHOLD BED_MAXTEMP // (°C) Skip user confirmation at or above this bed temperature (0 to disable)

    //#define POWER_LOSS_PIN             44 // Pin to detect power-loss. Set to -1 to disable default pin on boards without module, or comment to use board default.
    #define POWER_LOSS_STATE           HIGH // State of pin indicating power-loss
    //#define POWER_LOSS_PULLUP             // Set pullup / pulldown as appropriate for your sensor
    //#define POWER_LOSS_PULLDOWN

    //#define POWER_LOSS_ZRAISE        2    // (mm) Z axis raise on resume (on power-loss with UPS)
    #define POWER_LOSS_PURGE_LEN      20    // (mm) Length of filament to purge on resume

    // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
    // especially with "vase mode" printing. Set too high and vases cannot be continued.
    #define POWER_LOSS_MIN_Z_CHANGE    0.05 // (mm) Minimum Z change before saving power-loss data

    //#define BACKUP_POWER_SUPPLY           // Backup power / UPS to move the steppers on power-loss
    #if ENABLED(BACKUP_POWER_SUPPLY)
      #define POWER_LOSS_RETRACT_LEN     10 // (mm) Length of filament to retract on fail
    #endif

```

### Version of Marlin Firmware

latest bugfix-2.1.0

### Printer model

Prusa MK3S SKR BEAR

### Electronics

SKR 1.4 Turbo 1.4

### LCD/Controller

LCD 2004

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/16778184/config.zip)

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Dnts1, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ellensp, 2024-08-28:
Please ignore the hackers desperately trying to get you to download malicious code

ellensp, 2024-08-28:
real fix

```diff
diff --git a/Marlin/src/gcode/control/M80_M81.cpp b/Marlin/src/gcode/control/M80_M81.cpp
index e86e267313..ba5a0b6879 100644
--- a/Marlin/src/gcode/control/M80_M81.cpp
+++ b/Marlin/src/gcode/control/M80_M81.cpp
@@ -38,6 +38,10 @@
   #include "../../MarlinCore.h"
 #endif
 
+#if ENABLED(POWER_LOSS_RECOVERY)
+  #include "../../feature/powerloss.h"
+#endif
+
 #if ENABLED(PSU_CONTROL)
 
   /**
```

closed 2024-08-28 by thinkyhead