MarlinFirmware/Marlin #27377: [BUG] The abnormal behavior "PARK_HEAD_ON_PAUSE"

dupaser, 2024-08-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am using a BTT SKR PRO V1.2 and a MKS H43 DWIN display. 
At the moment I have version 2.0.9.5. I have tried using different versions of Marlin, there is a problem everywhere (somewhere the movement is worked out incorrectly sometimes the Watchdog is triggered).
When I activate `#define DGUS_LCD_UI_MKS`,  `#define ADVANCED_PAUSE_FEATURE` and `#define PARK_HEAD_ON_PAUSE`, the nozzle moves to the parking position but immediately returns to the last print location and then pauses. 
With St-Link debugging, it is visible that the next command is written to the "destination" variable after the coordinates of NOZZLE_PARK_POINT.

### Expected behavior

Stopping the nozzle at the NOZZLE_PARK_POINT

### Actual behavior

Stopping the nozzle at the NOZZLE_PARK_POINT and move to next coordinate 

### Steps to Reproduce

#define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2
#define DGUS_LCD_UI_MKS
#define USE_MKS_GREEN_UI
#define ADVANCED_PAUSE_FEATURE
#define PARK_HEAD_ON_PAUSE

### Version of Marlin Firmware

2.0.9.5

### Electronics

MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2

### LCD/Controller

DGUS_LCD_UI_MKS

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

dupaser, 2024-08-29:
1

dupaser, 2024-08-29:
Update info

thisiskeithb, 2024-08-29:
Please attach your [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) configs.

dupaser, 2024-08-29:
Good afternoon, thank you for your quick response. I downloaded your archive and configured it according to my parameters. There were issues with building the firmware in DGUS. Everywhere I made changes, I left a comment with the original value and an identifier "332211" so you could easily find them. The main problem was with the name `DGUS_SCREEN_MAIN`, which I replaced with `DGUSLCD_SCREEN_MAIN`, and `DGUS_SCREEN_STATUS`, which I replaced with `DGUSLCD_SCREEN_STATUS`. As a result, I ran my parameters and observed the same issues I described earlier. If the line is not commented out, the board restarts.

"Changes are only in /mks and in configuration files."

[](https://github.com/dupaser/Marlin-bugfix-2.1.x-1-/tree/main)

dupaser, 2024-08-30:
This is solution in void MKS_pause_print_move()

if (nozzle_park_mks.print_pause_start_flag) 
   { queue.exhaust(); }

narno2202, 2024-09-02:
@dupaser , I'm refactoring the MKS_H43 code when i saw your comment. I've already fixed some bugs and changed some wordings. Could you have a look?
https://github.com/narno2202/Marlin/tree/MKS_H43

closed 2024-12-16 by github-actions[bot]