MarlinFirmware/Marlin #27378: [BUG] FT_MOTION creates many resonance frequencies

vovodroid, 2024-08-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have gcode file moving axis back and forth with constant speed and decreasing amplitude, thus increasing frequency in 1..100 Hz range.

When 'printing' this file without FTM there are a few audible resonance peaks, main one can be canceled with IS. But there are many such peaks every 10-20 Hz or so with FTM.

I would attribute it to **FT_MOTION produces strange sound #27344.** Probably current pulsations induce resonance in many frequencies.

Though increasing FTM_FS, FTM_STEPPER_FS, FTM_STEPPERCMD_BUFF_SIZE increases pulsation frequencies, may be it still affects motion.

### Version of Marlin Firmware

https://github.com/MarlinFirmware/Marlin/commit/8ac0f1479648649c40873fecd9deec7396238f82

### Electronics

BTT SKR 3, TMC2209

### Bed Leveling

ABL Linear grid

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16798609/Marlin.zip)

thinkyhead, 2024-08-30:
Duplicate of #27344

There's only one actionable item in relation to this question, and we're still very much at the stage of collecting data and trying to understand how FT Motion behaves on our boards — especially faster boards — so that we can account for these extra periodic kicks and solve them. Let's continue to collect data and figure things out at the linked issue.

thinkyhead, 2024-08-30:
Duplicate of #27344

closed 2024-08-30 by thinkyhead