MarlinFirmware/Marlin #27384: Will Marlin support ANET v1.7 board? SD Card not working on default ANET A8 configurations.

vanepico, 2024-08-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Building 2.1.1 with all different possibilities of the SD settings (disable SD check, enable crc check, invert detect pin), the SD card refuses to mount (5% of the time it will automount if it is in when switched on)

### Bug Timeline

I am new to the firmware and it has never worked for me

### Expected behavior

When pressing attach media I expect the SD to be read and to be able to view contents

### Actual behavior

The system freezes for 5 seconds, and afterwards the SD remains unmounted.

### Steps to Reproduce

1. Put recommended bootloader onto v1.7 board
2. Take configuration and configuration ADV from Anet A8 folder and put in Marlin folder
3. Build 2.1.1/2.1.x bugfix and upload over USB
4. Insert SD card
5. In menu scroll down to attach media and press enter

### Version of Marlin Firmware

2.1.1.2 and 2.1.x bugfix

### Printer model

ANET A8 v1.7 board

### Electronics

Completely stock electronics apart from replacement motherboard

### LCD/Controller

Stock ANET A8

### Other add-ons

None yet

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-08-30:
As far we are aware the The Anet V1.0 V1.5 and V1.7 are all firmware identical. The changes are in power supply and protection  

That board doesn't have SD_DETECT_PIN, thus  why you need to manually attach media.   So "invert detect pin" is pointless 

There are a number of build environments sanguino1284p, sanguino1284p_optimized, melzi_optiboot
Which did you try?

ellensp, 2024-08-30:
I just tested this on my Anet 1.5
No issues 
sdcard mounted 10 out of 10 times 

You should reformat and/or replace you sdcard.

vanepico, 2024-08-30:
> As far we are aware the The Anet V1.0 V1.5 and V1.7 are all firmware identical. The changes are in power supply and protection
>
> That board doesn't have SD_DETECT_PIN, thus why you need to manually [...]

Thank you for the quick reply, that is good to know they are identical pin mapping.

I think the melzi_optiboot was the bootloader recommended somewhere, should I try sanguino bootloader?

I have tried reformatting the SD and tried other known working ones so I don't think this is the case.

vanepico, 2024-08-30:
If ANET boards do not have SD detect pin, why is `// Enable if SD detect is rendered useless (e.g., by using an SD extender)
//#define NO_SD_DETECT`

Not enabled in the ANET configs by default?

thisiskeithb, 2024-08-30:
> Not enabled in the ANET configs by default?

Because that printer doesn't ship with an SD extender.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server