MarlinFirmware/Marlin #27398: [BUG] undefined raise_after_is_relative on probe.h

jalcala993, 2024-09-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I just happened to be trying my first ever marlin build for my ender 3 and was hit with a build error introduced on the newest commit. 
[Configurations.zip](https://github.com/user-attachments/files/16842847/Configurations.zip)

Marlin\src\gcode\calibrate\../../module/probe.h:196:127: error: 'raise_after_is_relative' was not declared in this scope; did you mean 'raise_after_is_rel'?

indeed the variable was not properly refactored:

https://github.com/MarlinFirmware/Marlin/commit/5f0b9d429f6a7859bddbf592ac6eeb524ffcbb1d#diff-844640fa2c3026003779111f0d4594b9370678140e782a408409d57a9c1dc809

### Bug Timeline

https://github.com/MarlinFirmware/Marlin/commit/5f0b9d429f6a7859bddbf592ac6eeb524ffcbb1d

### Expected behavior

Build to complete. 

### Actual behavior

Build fails due to variable name mismatch. 

### Steps to Reproduce

1.- Download the latest code. 
2.- Attempt to build.
2.1 Extract the code
2.2 Open in visual studio
2.3 Install PlatformIO IDE
2.4 Attempt to build 

### Version of Marlin Firmware

bugfix-2.1.X

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

jalcala993, 2024-09-03:
@thinkyhead FYI, some unfortunate timing.

jalcala993, 2024-09-03:
There's another issue at 
https://github.com/MarlinFirmware/Marlin/commit/7aaba66d4dcc2ece2a3c7732f62b99aa1a0ba9e6#diff-0b1906f909a22c34a133c2ea6e89d6ed2a54b4694efe1acbafa7e10ad0cb3e55

Marlin\src\gcode\motion\G2_G3.cpp:319:7: error: 'thermalManager' was not declared in this scope

caused by removing 
#include "../../module/temperature.h"

thisiskeithb, 2024-09-03:
> There's another issue at [7aaba66#diff-0b1906f909a22c34a133c2ea6e89d6ed2a54b4694efe1acbafa7e10ad0cb3e55](https://github.com/MarlinFirmware/Marlin/commit/7aaba66d4dcc2ece2a3c7732f62b99aa1a0ba9e6#diff-0b1906f909a22c34a133c2ea6e89d6ed2a54b4694efe1acbafa7e10ad0cb3e55)
> 
> Marlin\src\gcode\motion\G2_G3.cpp:319:7: error: 'thermalManager' was not declared in this scope
> 
> caused by removing #include "../../module/temperature.h"

Fixed in https://github.com/MarlinFirmware/Marlin/commit/8c15a093fd2db82effa62944072b994be4bd3c0e

thisiskeithb, 2024-09-03:
CI is failing, so we're aware of the issue: https://github.com/MarlinFirmware/Marlin/actions/runs/10674977436/job/29586281708

Thinkyhead is working through these changes.

thisiskeithb, 2024-09-03:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/7cd56a57f96ad270419e84198deec4058bc0ec99

closed 2024-09-03 by thisiskeithb