MarlinFirmware/Marlin #27405: [BUG] FT_MOTION Stepper pauses an instant with intervals of about 1.2-1.3 seconds

marbocub, 2024-09-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using FT_MOTION, the stepper pauses an instant with intervals of about 1.2-1.3 seconds. It looks like that the pause is in sync with the update/redraw of the info screen on LCD. The faster the axis motion, the greater the shock by the pause.

A video is attached.

https://github.com/user-attachments/assets/6d186433-e09a-46a1-8770-116f4f21dd06

The vibration noise by the pause can be heard about every 1.2-1.3 seconds in the video. A "bump" caption is displayed when every pause occurs. It is also seen that the pause is in sync with the update of the info screen on LCD.

There are two round trips in the video. 60mm/s in the first round with 6 pauses and 40mm/s in the second round with 8 pauses. Thus the pause has nothing to do with the position on the axis.

### Bug Timeline

I found it yesterday but I don't know when the issue started.

### Expected behavior

I expected the stepper to move smoothly at a constant speed.

### Actual behavior

The stepper pauses an instant with intervals of about 1.2-1.3 seconds.

### Steps to Reproduce

1.Enable Fixed Time Motion in menu
2.Send motion gcode e.g. G1 X200 F3600
3.The issue is occurs

### Version of Marlin Firmware

bugfix-2.1.x (2024-09-05)

### Printer model

Sovol SV06 stock hardware

### Electronics

Stock electronics

### LCD/Controller

Sovol SV_NL_V1.3.1 controller (GD32F103RET6 and TMC 2209) with ENH12864Z-1 LCD, these are stock LCD/controller

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/user-attachments/files/16900776/Configurations.zip)

narno2202, 2024-09-06:
@marbocub , try to increase `FTM_STEPPERCMD_BUFF_SIZE` in `Configuration_adv.h`. The value can varie from printer to printer.

konradmb, 2024-09-12:
I've seen the same behaviour on GD32F103 since at least 2023-12 but without FT_MOTION and with COLOR_UI. Recently after lowering the DMA timings (as described in comment in `Marlin/src/HAL/STM32/tft/tft_fsmc.cpp`) the pauses got shorter, but still the bumps produce an audible scratch. Changing the menu to some static screen stops them.
It's like diring the DMA transfer to the screen, everything stops, even the ISR.

Nuck-TH, 2025-09-18:
This happens in COLOR_UI even without DMA(with TFT_SHARED_IO) at least on GD32F303. Affects both classic planner and FT motion.

ellensp, 2025-09-18:
From https://github.com/MarlinFirmware/Marlin/pull/25394  the PR that added FT_MOTION

"This feature uses a lot of SRAM for the step event buffer and does a lot of extra computation to apply shaping, so it requires a powerful board.

STM32F103 or GDGD32F103 are NOT powerful boards.  They don't even have hardware floating point units. (which I believe is a requirement)

oliof, 2026-01-09:
The original bug report referred to FT_MOTION from before @dbuezas ' full rewrite and the last update referring to code . So it would be good to know whether @marbocub sees this with current bugfix, since @ellensp 's comment in https://github.com/MarlinFirmware/Marlin/issues/27405#issuecomment-3309910780 still refers to the old implementation.

I know from my own experimentation that FT_MOTION runs on an SKR Pico, and I had it running on an STM32F0 board -- both without displays due to flash and RAM constraints. So my experience may not be relevant for this bug.

thinkyhead, 2026-09-01:
Please re-test with the latest version of FT Motion and if the timing issues still exist then we will dive into research to solve the issue asap. Once we have a detailed analysis we can allow @dbuezas _et. al._ to review the probable cause and make sure our proposed solution makes sense. GD32 timing has always been a tricky issue and we have discussed ways to improve base timing, so maybe this is a chance to improve in that general area as well.