MarlinFirmware/Marlin #27407: [BUG] Issues with Z-Offset when using BL-Touch (Marlin 2.1.2.4)

GER-HerrS, 2024-09-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a modified CR10-V2 with DirectDrive, BTT SKR2 board and BL-Touch.

I have had the problem I am describing here for a few versions (I can't remember exactly which one it was, I think 2.1.2.2).

At the time I had tried the current 2.1.0 bugfix and had no success with that either.

However, since it is not a new feature but a standard function, this should actually work and so I resignedly decided to wait for a new release version.

Now I am using 2.1.2.4 and the error has remained.

That makes me more than suspicious.

Maybe it is an error in my config files (although it worked like that in earlier versions?).

Maybe it is a bug after all.

The problem:
- I do auto-leveling with the BL-Touch (this works physically), the Z-axis moves to position 5.0mm
- I move the Z-axis down and notice that the nozzle is still over 5mm in the air at position 0 (according to the LCD)
- The Z-offset is -5.4mm according to the LCD
- I save the settings
- now I do a new auto-leveling, the Z-axis is automatically moved to position 5.0mm again, but everything is the same as the first time. The nozzle is again over 5mm above the bed at position 0. My Z-offset seems to have had no effect. But it is saved in the settings as -5.4.

[Marlin-Config.zip](https://github.com/user-attachments/files/16922658/Marlin-Config.zip)

### Bug Timeline

2.1.2.2

### Expected behavior

I expected BL-Touch Trigger point + Z-Offset to position the nozzle fine for first layer.

### Actual behavior

My Z-Offset seems to has no influence at all.

### Steps to Reproduce

Just try to use BL-Touch sensor with correct Z-Offset using Marlin 2.1.2.4.

### Version of Marlin Firmware

2.1.2.4

### Printer model

Creality CR-10 V2

### Electronics

BTT SKR2, Bondtech DDX v3 and Mosquito Hotend

### LCD/Controller

Standard LCD from CR-10 V2

### Other add-ons

Filament runout sensor

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

rondlh, 2024-09-08:
This should work fine, it's the basis of all bed leveling.
Try this:
1. Set the Z-offset to 0 (M851 Z0)
2. Heat the bed to the temperature you commonly use P140 S[your bed temp]
3. Heat the nozzle to 200 degrees (M104 S200)
4. Wait until the bed and nozzle have reached the set temperatures
3. Home the printer (G28) (move the center of the bed if needed)
5. Move the printer to Z=0 (G0 Z0)
7. Place a piece of thin paper on the bed, under the nozzle
8. Babystep until the nozzle starts grabbing the paper
9. Save your settings (M500)

Now your bed leveling should a roughly correct, some fine adjustments are usually still needed.

GER-HerrS, 2024-09-08:
I did it exactly like that (but directly on the controller).

But now I did it again using Octoprint and gcodes.

The result remains the same.

I took a photo and a video.

Pay attention to the Z position after the auto leveling is finished.

[image]

https://github.com/user-attachments/assets/4b41af18-130d-4c57-b737-a95400631d95

rondlh, 2024-09-09:
In German, nice :P

There are various parameters that affect the Z-position after homing, so the Z-position after homing doesn't tell you much.
Note that homing doesn't bring your Z-position to 0. Going to a Z=5mm is probably correct.

The display will always lag a bit, that's also not important. The 15.4 you see is probably the probe at it's highest point.
What is important is the response of M851 (probe offset), and where the nozzle is when you send the Z position to 0. (G0 Z0).
There is nothing going wrong in the video as far as I can see. That's all normal and correct.

GER-HerrS, 2024-09-09:
You got me :-D.

Just to make sure we're not talking past each other.
It shows 15.4mm while it's moving up the Z axis.
So it's moving to 15.4mm. But when it's reached it, it shows the position 5mm.
In fact, 15.4mm isn't right at all.
When I do the homing, I go down 5mm to get to position 0. Then I go down another 5.4mm until the nozzle scratches the paper.
So I'm expecting 10.4mm rather than 15.4mm.

In any case, something is wrong, because with the Z offset set, the nozzle is still too far above the bed after homing to position 0. A print would therefore print the first layer in the air.

GER-HerrS, 2024-09-11:
I have now specifically set the language to English and made two new videos.
The first video shows the normal auto homing process and that the nozzle is still well above the bed when position 0 has been reached.
Mind you: This is already with the Z offset of -5.4mm!

And again I can or must move the Z axis to position -5.4mm to grab the paper on the bed.
The printer seems to completely ignore the Z offset and so I can't print at all.
I don't know where this phenomenon comes from, but as I said, it has unfortunately not disappeared with the latest versions.

During the homing process, as can be seen in the last video, the Z position is 15.4mm. This is probably also controlled (5.4mm Z offset + 10mm nozzle lift from the bed).
But then the position is overwritten with 5mm without the Z axis actually moving to it.
So in reality it is at 15.4mm but according to the LCD it is at 5mm.
In my opinion there must be something wrong with the homing procedure?

Do you have any other ideas?

[image]

https://github.com/user-attachments/assets/76c73b5b-9436-4b78-99b4-eac9b7a0c4a9

https://github.com/user-attachments/assets/aa5ba29c-8a7f-4adb-971b-0ca22e984b03

Thanks

GER-HerrS, 2024-09-12:
No one able to help? Please get me printing again.

GER-HerrS, 2024-09-16:
Solution:
#define USE_ZMIN_PLUG
has to become:
//#define USE_ZMIN_PLUG

as I use no Z end stop except the bl touch.

closed 2024-09-16 by GER-HerrS