MarlinFirmware/Marlin #27410: [BUG] Build fails after https://github.com/MarlinFirmware/Marlin/pull/27338

oliof, 2024-09-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have an existing custom configuration for a machine that worked before https://github.com/MarlinFirmware/Marlin/pull/27338. As pointed out in https://github.com/MarlinFirmware/Marlin/pull/27338#issuecomment-2288202351, 27338 breaks previously working configs

### Bug Timeline

started with PR 27346

### Expected behavior

I expect my config.ini based configuration to continue to work.

### Actual behavior

build breaks

### Version of Marlin Firmware

2.1.-bugfix

### Printer model

RatRig V-Minion

### Electronics

SKR2

### LCD/Controller

Mellow 12864 Mini

### Other add-ons

TMC2209 steppers

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Since this is a build breakage, I am not including information from the printer. This issue persists including and up to commit id b94a0d036c843b59488a4de77c1de8cf6e9e5863

I have added my config.ini, and the generated Configuration.h Configuration_adv.h files in configs.zip and a buildlog from running `pio run -e STM32F407VG_btt_USB &> buildlog.txt`.

[configs.zip](https://github.com/user-attachments/files/16929213/configs.zip)
[buildlog.txt](https://github.com/user-attachments/files/16929243/buildlog.txt)

ellensp, 2024-09-10:
I cannot replicate this.

steps used:
Get a clean current copy of Marlin bugfix 2.1.x
extract config.ini  form provided configs.zip into Marlin directory  (same directory as Configuration.h)
Edit Config.ini  replacing the line
`ini_use_config                           = disable all`
with
`ini_use_config                           = all`

From a platformio new terminal run "pio run -e STM32F407VG_btt"  

This updates the Configuration.h and Configuration_adv.h files from config.ini and builds fine.

NOTE:

I notice in the provided Configuration.h from [configs.zip](https://github.com/user-attachments/files/16929213/configs.zip) has the line 
`#define HAS_MICROSTEPS true                 // Added by config.ini 2024-09-09 11:50:24
`

This is the cause of the build error listed, but this line is not in the Updated Configuration.h generated from the attached config.ini

thisiskeithb, 2024-09-11:
I don't use the `config.ini` method for configuring Marlin, but I confirmed that the provided configs fails when I dropped `config.ini`, `Configuration.h`, `Configuration_adv.h` onto a current copy of `bugfix-2.1.x`:

```prolog
Marlin/src/module/stepper.cpp: In static member function 'static void Stepper::microstep_init()':
Marlin/src/module/stepper.cpp:4102:46: error: 'MICROSTEP_MODES' was not declared in this scope
 4102 |     static const uint8_t microstep_modes[] = MICROSTEP_MODES;
      |                                              ^~~~~~~~~~~~~~~
*** [.pio/build/STM32F407VG_btt_USB/src/src/module/stepper.cpp.o] Error 1
==================================== [FAILED] Took 54.89 seconds ====================================

Environment          Status    Duration
-------------------  --------  ------------
STM32F407VG_btt_USB  FAILED    00:00:54.888
=============================== 1 failed, 0 succeeded in 00:00:54.888 ===============================
```

If I revert `Configuration.h` & `Configuration_adv.h` to stock and modify `config.ini` as ellensp noted above, that `#define HAS_MICROSTEPS true` line is not added and the firmware builds fine.

So why/how does that `#define HAS_MICROSTEPS true` line get added in on OP's build? I don't know.

On our [Configuring with INI page](https://marlinfw.org/docs/configuration/config-ini.html), there's a note that:

> the behavior of this feature is likely to change as we continue to experiment and find new ways to use it, so consult the codebase for the most up to date information.

So it's really still a beta feature and quite brittle since there's no error checking/cleaning of old values & you'll have to regenerate a working `config.ini` using a current clean config (that hasn't been previous generated/modified by build scripts).

thinkyhead, 2024-12-03:
Automated migration tools will be incorporated into Auto Build Marlin pretty soon. In the meantime you can use our [community resources](https://www.patreon.com/posts/new-reward-117183843?utm_medium=clipboard_copy&utm_source=copyLink&utm_campaign=postshare_creator&utm_content=join_link) to get help with updating configurations to a newer version.

closed 2024-12-03 by thinkyhead