MarlinFirmware/Marlin #27412: [BUG] MMU3 Enabled - MMU3 Filament load error Cant Resume from Parking

SpankyDevil, 2024-09-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If MMU3 throws an error during Printing for example "FSENSOR DIDN'T TRIGG." The Extruder goes to Parking Position and the screen displays Retry or More Info.

If you don't catch this error straight away the Hotend and Bed cooldown then when you click to Retry, The MMU3 starts to Load the Filament and causes Marlin to reboot.

If you have Power Feature enabled, Marlin will prompt to Resume Print or Cancel
If you Resume Nothing Happens. 

### Bug Timeline

when MMU3 feature was merged

### Expected behavior

Click Retry,
Extruder and Bed heat up then MMU3 Loads Filament and Print Continues

### Actual behavior

As Above in Description

### Steps to Reproduce

Trigger MMU3 Error, wait till Hotend and Bed cooled down then retry.

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Sep  3 2024 09:30:57)

### Printer model

Prusa MK3S+ Clone

### Electronics

SKR 3, TMC2209

### LCD/Controller

BTT TFT50 V3.0

### Other add-ons

Prusa MMU3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16937229/Marlin.zip)

thisiskeithb, 2024-09-09:
Duplicate of #27396

closed 2024-09-09 by thisiskeithb

SpankyDevil, 2024-09-09:
This is a different issue to https://github.com/MarlinFirmware/Marlin/issues/27396
Please reopen

thisiskeithb, 2024-09-09:
As I mentioned on Discord, the core issue is cold loading causing rebooting. This duplicates the other open bug report.

SpankyDevil, 2024-09-09:
I'm afraid I have to disagree, The issue is Marlin did not heat the hotend before performing the MMU3 Retry.