MarlinFirmware/Marlin #27413: [BUG] MMU3 Enabled - MMU3 Filament load error, Retry doesn't Heat Hotend before Retying

SpankyDevil, 2024-09-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If MMU3 throws an error during Printing for example "FSENSOR DIDN'T TRIGG." The Extruder goes to Parking Position
If you have #define HOTEND_IDLE_TIMEOUT enabled Hotend cools down.
the screen displays Retry or More Info.

If you don't catch this error straight away the Hotend and Bed cooldown then when you click to Retry, The MMU3 starts to Load the Filament cold.

### Bug Timeline

when MMU3 feature was merged

### Expected behavior

Click Retry,
Extruder and Bed heat up then MMU3 Loads Filament and Print Continues

### Actual behavior

As Above in Description

### Steps to Reproduce

Trigger MMU3 Error, wait till Hotend and Bed cooled down then retry.

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Sep 3 2024 09:30:57)

### Printer model

Prusa MK3S+ Clone

### Electronics

SKR 3, TMC2209

### LCD/Controller

BTT TFT50 V3.0

### Other add-ons

Prusa MMU3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16938799/Marlin.zip)

thisiskeithb, 2024-09-10:
In your original issue (#27412), you reported that:

> If you don't catch this error straight away the Hotend and Bed cooldown then when you click to Retry, The MMU3 starts to Load the Filament and **causes Marlin to reboot**

Does Marlin no longer reboot when trying to load filament cold?

thisiskeithb, 2024-09-10:
Duplicate of #27396

thisiskeithb, 2024-09-10:
Please use #27396 for any and all MMU3-related + Marlin rebooting when cold issues

closed 2024-09-10 by thisiskeithb