MarlinFirmware/Marlin #27419: [BUG] Z axis does not move after first couple of layers

thk4711, 2024-09-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I start printing everything is fine for the first couple of layers. Then the printer will continue to do the x and y movements but does not move the z axis anymore. I am not sure - but to me it looks like it stops when it has reached the value of the z offset of my BLTOUCH.
[Configuration.zip](https://github.com/user-attachments/files/16997433/Configuration.zip)

### Bug Timeline

2024-09-13

### Expected behavior

The printer should move the z axis until the print is finished

### Actual behavior

It stops after the first couple of layers

### Version of Marlin Firmware

bugfix-2.1.x (2024-09-13)

### Printer model

Creality CR10S

### Electronics

Bigtreetech SKR MINI E3 V3.0

### LCD/Controller

Bigtreetech TFT35-E3

### Other add-ons

BLTOUCH sensor

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-09-14:
I cannot replicate this on a test jig, matched motherboard and LCD 

I 'printed" up to Z10, and Z moved as expected

NOTE:
 I had to make minor changes to your provided config 
#define TEMP_SENSOR_0 998 // fake thermistor, gcode to heat hotend removed, E set to extrude cold so no warning etc.
#define TEMP_SENSOR_BED 998  // fake thermistor, gcode to heat bed removed.
#define FIX_MOUNTED_PROBE vs #define BLTOUCH 
#define X_MIN_ENDSTOP_HIT_STATE LOW
#define Y_MIN_ENDSTOP_HIT_STATE LOW
#define Z_MIN_ENDSTOP_HIT_STATE LOW
#define Z_MIN_PROBE_ENDSTOP_HIT_STATE LOW

I would check all your mechanical stuff. In particular for cables or bits of filament stuck in your Z leadscrew.

thk4711, 2024-09-14:
Thank You for your response. I don’t think there is a mechanical or wiring issue causing my problem. Everything was working with the original creality mainboard. I used the same version of Marlin. I also downloaded a pre-built firmware for my new mainboard from bigtreetech(it was for an Ender 3 so the dimensions were wrong). [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin)
With that version printing was OK. After that I built Marlin-lts-2.0.9 for my printer but also without success. So far I was not able to find out what is different on the bigtreetech firmware compared to what I am using.

classicrocker883, 2024-09-14:
https://github.com/MarlinFirmware/Marlin/issues/26323

this one sounds related

thk4711, 2024-09-17:
Since I had no Idea what to do here I started from scatch with a new configuration for my Creality/CR-10S/BigTreeTech SKR Mini E3 3.0 combination. Somehow it is working now but I have no idea why. I made a diff of the Configuration.h working and not working and the same for the Configuration_adv.h 
Maybe it will help someone later to find out, what is going on here.

[MarlinConfigDiff.zip](https://github.com/user-attachments/files/17028177/MarlinConfigDiff.zip)

closed 2024-09-17 by thisiskeithb