MarlinFirmware/Marlin #27429: Ender3 V3 SE build failing with `One or more timer conflict detected`

leonardola, 2024-09-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

While compiling v2.1.2.4 for my Ender 3 V3 SE I get the following error `One or more timer conflict detected. Examine \"timers_in_use\" to help identify conflict.`

I've merged the config files from [creality official repo](https://github.com/CrealityOfficial/Ender-3V3-SE) and the [default 2.1.2.4 config files](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x/config#:~:text=https%3A//github.com/MarlinFirmware/Configurations/archive/bugfix%2D2.1.x.zip).

[Creality Configuration.h](https://github.com/CrealityOfficial/Ender-3V3-SE/blob/main/Marlin/Configuration.h)
[Creality Configuration_adv.h](https://github.com/CrealityOfficial/Ender-3V3-SE/blob/main/Marlin/Configuration_adv.h)

I also added the board [pins file](https://github.com/CrealityOfficial/Ender-3V3-SE/blob/main/Marlin/src/pins/stm32f1/pins_CREALITY_V3_GD303.h), registered the board in pins.h and boards.h

I did find other issues about this error message but I still can't figure it out.
Any ideas?

### Steps to Reproduce

- Checkout to [this PR](https://github.com/MarlinFirmware/Marlin/pull/27428) branch
- Click build on the Auto Build Marlin tab
- Notice the error

### Version of Marlin Firmware

2.1.2.4

### Printer model

Creality Ender-3 V3 SE

### Electronics

All Stock

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[creality-marlin.zip](https://github.com/user-attachments/files/17078270/creality-marlin.zip)

ellensp, 2024-09-21:
Issue makes no sense.

You talk about 2.1.2.4 but configs and your github are bugfix.

Provided configs are just a mess...

This specific error 

This is wrong.

 `#include "stm32f1/pins_CREALITY_V3_GD303.h"       // STM32F1                              env:STM32F103RET6_creality`

In renamed.ini
[env:STM32F103RET6_creality]          ;=> STM32F103RE_creality
extends = renamed

ie this was renamed and shortened, in Marlin it should be 

 `#include "stm32f1/pins_CREALITY_V3_GD303.h"       // STM32F1                              env:STM32F103RE_creality`

thisiskeithb, 2024-09-21:
Some more backstory: This user also joined Discord earlier today and I eventually worked out this board was already supported (`BOARD_CREALITY_CR4NS` added in https://github.com/MarlinFirmware/Marlin/pull/27003), so:

- Pins still need to be updated (https://github.com/MarlinFirmware/Marlin/pull/27003#issuecomment-2235358590)
- Configs need to be updated to current `bugfix-2.1.x`
- Optionally: Port the rest of Creality’s changes upstream like their WiFi and leveling / whatever else they changed in the V3 SE fork.

ellensp, 2024-09-21:
[Configuration.zip](https://github.com/user-attachments/files/17083009/Configuration.zip)

Updated, real configs that build

I have not addressed the board duplication issue.

closed 2024-09-21 by ellensp