MarlinFirmware/Marlin #27430: Canceled objects start to print again after pause.

how-eee, 2024-09-21:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Canceled objects start to print again after pause.

[Configs.zip](https://github.com/user-attachments/files/17084398/Configs.zip)

### Steps to Reproduce

1. Start print file with multiple objects marked M864 Sxxx
2. Cancel some objects on lcd menu
3. Pause print via lcd menu
4. resume print.

### Version of Marlin Firmware

2.1.2.4

### Printer model

kossel delta

### Electronics

skr 1.3

### LCD/Controller

discount 

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-09-21:
> ### Did you test the latest `bugfix-2.1.x` code?
> 
> No, but I will test it now!

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

how-eee, 2024-09-21:
> > ### Did you test the latest `bugfix-2.1.x` code?
> > No, but I will test it now!
> 
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Latest bugfix with newly filled config. Issue still there.

closed 2025-01-10 by github-actions[bot]