MarlinFirmware/Marlin #27431: Homing failed Printer Halted due to delta height plus probe offset

how-eee, 2024-09-21:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Probe offset = -15mm
G33 calculated delta heiht  = 285mm
So full Z travel 300mm. Printer halted at ~290mm of homing.

Should I manually set Probe offset = 0mm, delta heiht  = 300mm, homing OK

[Configs.zip](https://github.com/user-attachments/files/17084409/Configs.zip)

### Version of Marlin Firmware

2.1.2.4

### Printer model

kossel delta

### Electronics

skr 1.3

### LCD/Controller

discount 

### Bed Leveling

No Bed Leveling

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-09-21:
> ### Did you test the latest `bugfix-2.1.x` code?
> 
> No, but I will test it now!

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

how-eee, 2024-09-21:
> > ### Did you test the latest `bugfix-2.1.x` code?
> > No, but I will test it now!
> 
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

I cant compile firmware bin with my config. Configuration.h becomes blank

thisiskeithb, 2024-09-21:
> Configuration.h becomes blank

Which operating system & version are you running?

Can you post your compile log?

how-eee, 2024-09-21:
I was forced to take config files from bugfix folder and made all again. Waiste1 hour. But I found new bug there.

  //#define SHOW_TEMPERATURE_BELOW_ZERO
An error occured if uncomment that.

how-eee, 2024-09-21:
OK. Thank you to show me reason why should I use bugfix branch instead of release one.

NOW
Probe offset = -15mm
G33 calculated delta heiht = 285mm
So full Z travel 300mm. Printer NOT halted. Homed correctly.

Issue solved!
Thanks again.

thisiskeithb, 2024-09-21:
> Issue solved!
> Thanks again.

Great!

closed 2024-09-21 by thisiskeithb