MarlinFirmware/Marlin #27438: [BUG] Babystepping not working on BTT SKR Mini E3 V3.0

blu28, 2024-09-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The screen that says "Babystepping Probe Z" displays the current probe Z value, but turning the knob does nothing, it does not change the value. Pressing the knob take you back to the previous menu.

### Bug Timeline

It has never worked for me. I first started trying this in July.

### Expected behavior

I expect the number to increment and decrement as the knob is turned left and right.

### Actual behavior

The number did not change.

### Steps to Reproduce

1. Go to the "Babystepping Probe Z" screen.
2. turn the knob.

### Version of Marlin Firmware

2.1.x branch as of 2024/09/23

### Printer model

Ender-3 Pro

### Electronics

BTT SKR Mini E3 V3.0

### LCD/Controller

Stock controller

### Other add-ons

BLTouch clone.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

blu28, 2024-09-23:
[Configuration.zip](https://github.com/user-attachments/files/17097493/Configuration.zip)

blu28, 2024-10-11:
Making the BABYSTEP_MULTIPLICATOR_Z set to 1 causes the BABYSTEP to respond.

ellensp, 2024-10-11:
//#define BABYSTEP_MILLIMETER_UNITS       // Specify BABYSTEP_MULTIPLICATOR_(XY|Z) in mm instead of micro-steps
  #define BABYSTEP_MULTIPLICATOR_Z  0.01       // (steps or mm) Steps or millimeter distance for each Z babystep

you cannot have steps < 1  when your not in BABYSTEP_MILLIMETER_UNITS mode

closed 2024-10-11 by ellensp