MarlinFirmware/Marlin #27440: [BUG] Commit "Skip bad compilers" breaks Windows build

vovodroid, 2024-09-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After commit "Skip bad compilers" https://github.com/MarlinFirmware/Marlin/commit/f2d585ac7f5dad29aa57112ea327812f91924088 Windows build in Visual Code fails:

```
PS C:\DISK_D\3D\Marlin> platformio run -e STM32H723VG_btt
Processing STM32H723VG_btt (board: marlin_STM32H723VG; platform: ststm32@15.4.1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
AttributeError: module 'os' has no attribute 'uname':
  File "C:\Users\User\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\User\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\User\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 316:
    apply_features_config()
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 136:
    if not env.MarlinHas(feature):
  File "C:\Users\User\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 283:
    load_marlin_features()
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 271:
    define_list = run_preprocessor(env)
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\preprocessor.py", line 27:
    cxx = search_compiler(env)
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\preprocessor.py", line 79:
    sysname = os.uname().sysname
======================================================================================= [FAILED] Took 1.02 seconds =======================================================================================
Environment      Status    Duration
---------------  --------  ------------
STM32H723VG_btt  FAILED    00:00:01.018
=
```

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/17114989/Marlin.zip)

ellensp, 2024-09-24:
looks like Linux specific code os.uname() was intraduced  in https://github.com/MarlinFirmware/Marlin/commit/f2d585ac7f5dad29aa57112ea327812f91924088#diff-1e028fbdef12eea0e3e0913d3d18c0310fa76aa53750abe299585807cdca4ddcR79

https://docs.python.org/3/library/os.html#os.uname  says "Availability: Unix."
so it is not available on windows

andrademf, 2024-09-24:
Same issue here. I could fix it temporarily by changing this:
[image]

ellensp, 2024-09-25:
With https://github.com/MarlinFirmware/Marlin/commit/9d93ad7759cfeb1a556825dfe14e4b937b8d81b4  os.uname()  has been removed.

closed 2024-09-25 by ellensp

thisiskeithb, 2024-09-25:
> With [9d93ad7](https://github.com/MarlinFirmware/Marlin/commit/9d93ad7759cfeb1a556825dfe14e4b937b8d81b4) os.uname() has been removed.

Now https://github.com/MarlinFirmware/Marlin/commit/e2c0150fa82a9d3e730a28236a4b83a3a408e12b since there was a force-push.

In all cases, grab the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) for the fix.