MarlinFirmware/Marlin #27444: [BUG] Wrong home position after applying home offset

Palpalych13, 2024-09-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I had own configuration of Marlin with Voxelab Aquila C2 + BTT mini mainboard.
I configure my printer to sensorless homing and want to apply some home offsets, but I gave some troubles after that.

My investigation shown that problem, After applying that settings, the printer do home parking incorrect. After the attempt of X-parking, my printer stops further positioning of the Y axis, and home coordinates not right to start printing.

The problem I has solved by update a string in the function home_z_safely() in the 147 line of G28.cpp module.
At past was be addressing to default function with 0 mm/s feedrate, which finished with abnormal result after X-homing when applying home offsets.

Wrong string was: do_blocking_move_to_xy(destination);
The new one: do_blocking_move_to_xy(destination, 10);

[image]

### Bug Timeline

Probably the old

### Version of Marlin Firmware

2.1.2.4

### Printer model

Any (but I has Voxelab Aquila C2 + BTT mini mainboard)

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2024-10-09:
The feedrate for this move is determined by `XY_PROBE_FEEDRATE_MM_S`, which will come from the best source it can determine, like so:
```cpp
#if HAS_ABL_NOT_UBL
  extern feedRate_t xy_probe_feedrate_mm_s;
  #define XY_PROBE_FEEDRATE_MM_S xy_probe_feedrate_mm_s
#elif defined(XY_PROBE_FEEDRATE)
  #define XY_PROBE_FEEDRATE_MM_S MMM_TO_MMS(XY_PROBE_FEEDRATE)
#else
  #define XY_PROBE_FEEDRATE_MM_S PLANNER_XY_FEEDRATE()
#endif
```
Check your settings and adjust your homing and probing feedrates and see if you can fix the issue without imposing a 10 mm/s feedrate on all configurations.

Palpalych13, 2024-10-19:
The XY_PROBE_FEEDRATE_MM_S not used in Configuration.h.
Is the NOZZLE_PARK_XY_FEEDRATE but it can be undefined.
I decided to fix this problem like this:
[image]

Palpalych13, 2024-10-20:
Not actually now. Revised this issue. The system reset was caused by moving the nozzle set in XY_PROBE_FEEDRATE too fast (133 mm/s in my case). At the same time, I used an increased number of microsteps for steppers XY.

closed 2024-10-20 by Palpalych13