MarlinFirmware/Marlin #27446: [BUG] Display issues on Ender 3 V2 Neo after firmware update

albka1986, 2024-09-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After flashing the version bugfix 2.1 Marlin firmware to my Ender 3 V2 Neo, the screen became corrupted (see attached photo). The UI elements are distorted, with unreadable text and misaligned graphics. The issue occurred immediately after the firmware update. No specific files for the display were included in the firmware package I used, so I suspect this could be related to a missing or incorrect screen firmware configuration. 
[image]

Attached also photo of display: 
[image]

### Steps to Reproduce

Steps to reproduce:

Download the latest Marlin firmware "bugfix 2.1" for Ender 3 V2 Neo.
Flash the firmware to the printer.
Reboot the printer and observe the display.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender-3 V2 Neo

### Electronics

board 4.2.2

### LCD/Controller

not sensor, with button  (attached photo)

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/17179085/Marlin.zip)

thisiskeithb, 2024-09-29:
Give the Ender-3 V2 LCD firmware a try. Full instructions are in the README: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2

> [!NOTE]  
> The TJC display variant is undocumented and incompatible with custom UIs.

albka1986, 2024-09-29:
@thisiskeithb.
When I upload the 2 folders to the mSD card, the display does not update. Nothing happens. The firmware is not updated.
[image]

ellensp, 2024-09-29:
TJC uses TJC_SET/tjc.tft. We cannot generate those files, so there is no screen updated for this display provided by Marlin.

thisiskeithb, 2024-09-29:
> TJC display (ie what you have) uses TJC_SET/tjc.tft. We cannot generate those files, so there is no screen updates for this display provided by Marlin.

Ahh. I interpreted the "Currently incompatible with custom UIs" / "there are currently no custom display firmwares for this unit" wording as `DWIN_CREALITY_LCD` being the only supported UI, but that doesn't seem to be the case.

I'll update the README to clarify that TJC displays are unsupported and users will have to go headless or swap in a supported display.

thisiskeithb, 2024-09-29:
TJC display support has been clarified in https://github.com/MarlinFirmware/Configurations/pull/1105

`DWIN_MARLINUI_PORTRAIT` would be your best bet if you want to use a TJC display, but we have no way to compile/add graphic assets for them.

closed 2024-09-29 by thisiskeithb

albka1986, 2024-10-06:
I updated the line, but still see the issue with displaying icons:
[image]
[image]