MarlinFirmware/Marlin #27452: [BUG] Printer hangs mid print on latest bugfix

dwisnie2, 2024-10-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was trying to get the input shaper working, on 2.1.2.4 but my Y axis was losing steps after ~2 minutes of printing at any speed and acceleration (exactly the same problem and hardware as here: https://github.com/MarlinFirmware/Marlin/issues/27257). After trying the 2.1.x bugfix the issue was gone, but the printer is slowing down by much during acceleration and randomly hangs mid-print even with input shaping turned off. The LCD and menu still work, but trying to stop printing makes no effect (prompt does not disappear). I tried printing from the host and SD, but got the same effect. I do not have a hanging problem on 2.1.2.4.
[Configuration.zip](https://github.com/user-attachments/files/17235826/Configuration.zip)

### Expected behavior

printing

### Actual behavior

not printing

### Version of Marlin Firmware

8c61606691a52298236811b6cbacc965d8c1f9dc

### Printer model

Ender 3

### Electronics

SKR 1.3

### LCD/Controller

stock ender

### Other add-ons

TMC 2208

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

benjmharless, 2024-11-20:
I have been having a similar issue where the printhead freezes but the heaters stay on and the screen still works and acts like it is still printing. This is a Ender 3 with SKR 1.4 turbo, TMC2209 drivers, running Marlin 2.1.2.

benjmharless, 2024-12-04:
I fixed my issue by enabling RX_BUFFER_SIZE and TX_BUFFER SIZE and setting them to the highest option, and by changing the serial ports to match this: https://user-images.githubusercontent.com/54359396/132397091-d596abcf-750f-422a-bb59-afafc246dc58.jpg
I am now running Marlin 2.1.2.5.
Hope that helps!

closed 2025-03-19 by github-actions[bot]

thinkyhead, 2025-03-27:
I would have guessed that the TMC drivers were overheating and going into standstill mode, which we can sometimes but not always detect by sending a request to read the TMC driver info.

But that had nothing to do with your issue, as it turns out?

There is nothing in Marlin code that stops sending signals to the stepper drivers, and if interrupts generally broke then temperatures would stop updating too. That's why I lean towards the problem being related to overheating stepper drivers. But then… would all axes stop?

Well, it is a conundrum. When an issue can be reproduced quickly and easily by running some test on your end, then we can use tricky methods to track down breaking changes in the code by testing the firmware at different points in time. But if the issue is spontaneous and takes a long time to manifest, it can be a bigger challenge to locate and fix the cause of issue.

benjmharless, 2025-03-30:
I don't think it was the drivers because changing the serial ports fixed it.